ros2/rclcpp #393: wrapper object use after free

gaoethan, 2017-11-09:
## Bug report

**Required Info:**

- Operating System: 
  -  N/A
- Installation type: 
  -  N/A
- Version or commit hash:
  -  master
- DDS implementation: 
  -  N/A
- Client library (if applicable): 
  -  rclcpp

#### Steps to reproduce issue

It occurs the issue of wrapper object use after free with `c_str()` in the code snippet like the following in [transition.cpp](https://github.com/ros2/rclcpp/blob/master/rclcpp_lifecycle/src/transition.cpp#L45-L55):
  ```
auto start_state = new rcl_lifecycle_state_t;
  start_state->id = start.id();
  start_state->label = start.label().c_str();

  auto goal_state = new rcl_lifecycle_state_t;
  goal_state->id = goal.id();
  goal_state->label = start.label().c_str();

  transition_handle->start = start_state;
  transition_handle->goal = goal_state;
```
because The internal representation of temporary of type `std::__cxx11::string` escapes into `start_state->label`, but is destroyed when it exits scope, I intend to draft a PR to optimize it, for example, using s'th like strcpy() to keep it firstly and then give that to `start_state->label`, but it involves some other change and maybe it's better check more comments before getting down to modify it. any comments ? thanks !

#### Expected behavior
the access to content should be at valid address
#### Actual behavior
use after free
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

Karsten1987, 2017-11-09:
Good catch.
It might be worth checking into returning a `const char *` for the `label()` function if you intend on going with `strcpy()`

I'd propose you file a PR for this ticket so we can run CI on it.

Karsten1987, 2017-12-06:
closed via #418

closed 2017-12-06 by Karsten1987