ros2/rclcpp #394: Missing copy constructor and assignment operator

gaoethan, 2017-11-09:
## Bug report

**Required Info:**

- Operating System:
  -  N/A
- Installation type:
  -  source
- Version or commit hash:
  -  master
- DDS implementation:
  -  N/A
- Client library (if applicable):
  -  rclcpp

#### Steps to reproduce issue

In the following design of [state.cpp](https://github.com/ros2/rclcpp/blob/master/rclcpp_lifecycle/src/state.cpp#L47-L58):
```
State::State(const rcl_lifecycle_state_t * rcl_lifecycle_state_handle)
: owns_rcl_state_handle_(false)
{
  state_handle_ = rcl_lifecycle_state_handle;
}

State::~State()
{
  if (owns_rcl_state_handle_) {
    delete state_handle_;
  }
}

```
Class `rclcpp_lifecycle::State` owns resources that are freed in its destructor but has no user-written copy constructor or assignment operator. The destructor frees member state_handle_ when `owns_rcl_state_handle_` is true. but the class is copy-constructed and copy-assigned the the following code snippet, for example:

[lifecycle_node_interface_impl.hpp #261](https://github.com/ros2/rclcpp/blob/master/rclcpp_lifecycle/src/lifecycle_node_interface_impl.hpp#L261)
[lifecycle_node_interface_impl.hpp #356](https://github.com/ros2/rclcpp/blob/master/rclcpp_lifecycle/src/lifecycle_node_interface_impl.hpp#L356)

That's to say this class, which frees resources in its destructor, does not have a user-written copy constructor, but is copy-constructed. Unless the resource is managed separately from the copy operation, this may cause use-after-free errors. In addition, the same case to the design in [transition.cpp](https://github.com/ros2/rclcpp/blob/master/rclcpp_lifecycle/src/transition.cpp#L59-L76)

We know that their `owns_rcl_state_handle_` and `owns_rcl_transition_handle_` defaults to false, so..here to confirm that these were designed like this on purpose? and will it be prone to introduce new issues in the future while using ? thanks !

Karsten1987, 2017-11-09:
That's definitely a missing feature! Thanks for catching this. 
I also propose you open a PR so we can run CI on it.

Karsten1987, 2017-12-06:
closed via https://github.com/ros2/rclcpp/pull/419

closed 2017-12-06 by Karsten1987