ros2/rclcpp #398: Build failed on macOS platform after SHA a215d2d22e0328

minggangw, 2017-11-17:
## Bug report

**Required Info:**

- Operating System:
  - macOS with xcode 8.3.3
- Installation type:
  - build from source
- Version or commit hash:
  - a215d2d22e0328
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Build ROS2 by instruction from https://github.com/ros2/ros2/wiki/OSX-Development-Setup

#### Expected behavior
Can build successfully

#### Actual behavior
Build failed
```
/Users/distiller/ros2_ws/src/ros2/rclcpp/rclcpp/src/rclcpp/duration.cpp:130:28: error: 
      no member named 'abs' in namespace 'std'; did you mean simply 'abs'?
  auto abs_lhs = (uint64_t)std::abs(lhsns);
                           ^~~~~~~~
                           abs
/usr/include/stdlib.h:129:6: note: 'abs' declared here
int      abs(int) __pure2;
         ^
/Users/distiller/ros2_ws/src/ros2/rclcpp/rclcpp/src/rclcpp/duration.cpp:131:28: error: 
      no member named 'abs' in namespace 'std'; did you mean simply 'abs'?
  auto abs_rhs = (uint64_t)std::abs(rhsns);
                           ^~~~~~~~
                           abs
/usr/include/stdlib.h:129:6: note: 'abs' declared here
int      abs(int) __pure2;
         ^
/Users/distiller/ros2_ws/src/ros2/rclcpp/rclcpp/src/rclcpp/duration.cpp:158:28: error: 
      no member named 'abs' in namespace 'std'; did you mean simply 'abs'?
  auto abs_lhs = (uint64_t)std::abs(lhsns);
                           ^~~~~~~~
                           abs
```

Log from https://circleci.com/gh/minggangw/rclnodejs-1/294
#### Additional information
add ?
``` 
#include <cstdlib> 
```

mikaelarguedas, 2017-11-17:
Thanks @minggangw for reporting.
You are totally right, the header is missing, surprised cpplint didnt complain about the missing header.
 #399 should fix this

closed 2017-11-17 by mikaelarguedas

minggangw, 2017-11-19:
@mikaelarguedas thanks for fixing it quickly 👍

PR #399 (2017-11-17): include cstdlib for std::abs function
fixes #398 

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]