ros2/rclcpp #409: Can't create ParameterService from within node constructor

dhood, 2017-11-27:
Working on https://github.com/ros2/demos/pull/192 I realised that classes that inherit from Node don't have an appropriate constructor they can call (from what I can see) to create a parameter service.

The only ParameterService constructor there is [takes a node shared pointer](https://github.com/ros2/rclcpp/blob/070b3125c1a7f76fbde3e31973d89a052a2ea652/rclcpp/include/rclcpp/parameter_service.hpp#L44), which can't be used from within a Node constructor because its weak pointer won't have been initialised yet. From a usability standpoint this means the parameter service needs to be created after the node is instantiated, not in its constructor, which is not ideal.

@Karsten1987 mentioned that this was at one point an issue for regular Clients/Services as well but was addressed by https://github.com/ros2/rclcpp/commit/f0afcabf052161c7b44b10665e74e34593dc55bf (see https://github.com/ros2/rclcpp/pull/277#discussion_r91813549).

There's a workaround available for parameter clients by using [the AsyncParametersClient constructor that takes the interface pointers](https://github.com/ros2/rclcpp/blob/022b2b1b807d8239b492e81a8fc156e36c60724d/rclcpp/include/rclcpp/parameter_client.hpp#L52), as the interfaces will have been initialised by that point. However, the call will look like:
```
parameters_client_ = std::make_shared<rclcpp::parameter_client::AsyncParametersClient>(
  get_node_base_interface(),
  get_node_topics_interface(),
  get_node_graph_interface(),
  get_node_services_interface());
```
which is not a great user experience either.

---

Would it make sense to:
1. Have a constructor for the ParameterService that takes the relevant interfaces, and the version that takes the node shared pointer will just forward to it (as is done for AsyncParametersClient)? This will allow users to create a parameter service in the constructor, but it will not be pretty.
1. Make convenience functions in the Node class for creating parameter services and parameter clients, so that the work of getting the node interfaces (shown above) is hidden from users?

Neither of these changes are necessary if users instead do something like:
```
auto node = std::make_shared<ParameterNode>();
auto parameters_client_ = std::make_shared<rclcpp::parameter_client::AsyncParametersClient>(node);
auto sub = parameters_client_->on_parameter_event(std::bind(&ParameterNode::on_parameter_event, node, std::placeholders::_1));
```

or if the node has a post-constructor initialisation step. I don't think either of those should be necessary, however the parameter clients/services seem to fundamentally be treated differently to normal services (e.g. that you have to pass a node into the constructor, can anyone point me to why this is?) so perhaps their expected usage is different as well. I think it's actually just something we haven't gotten around to yet, but want to make sure.

Karsten1987, 2017-11-27:
> I don't think either of those should be necessary, however the parameter clients/services seem to fundamentally be treated differently to normal services (e.g. that you have to pass a node into the [...]

I can't tell why they are differently, but I think it's a good idea to expose a function in the node class similar to regular services, such as `node->create_parameter_<service|client>` and internally forward this to the `node_parameter_interface`. This goes along with the current implementation of pubs, subs and srvs and therefore can be called from the constructor.

wjwwood, 2017-11-27:
What @Karsten1987 suggested is ok, but we should also be able to have a constructor signature for the parameter client/service classes which take `Node *` and then get the interface pointers and pass them along to the more complicated constructor, that way you can pass `this` in the constructor.

We can do both things.

dhood, 2017-11-28:
> we should also be able to have a constructor signature for the parameter client/service classes which take `Node *` and then get the interface pointers and pass them along to the more complicated constructor, that way you can pass `this` in the constructor.

Would we expect someone to pass `this` (a `Node *`) into the constructor of a parameter client, though? I didn't think we wanted to support that. It's not supported for other entities e.g. non-parameter clients, but I didn't feel like it's "still todo", but rather because we want users to call `create_*` methods on the node instead of calling the constructors directly.

The idea I had with the suggestion of adding the equivalent `create_*` method on the node for parameter clients/services (pt 2) was in order to move _away_ from passing a node into the parameter service/client constructor, and probably even remove the constructor that's there now. This sounds like it's going in the other direction to what you're saying though :smile:

wjwwood, 2017-11-28:
You may be right about not wanting to use `Node *`, I'd have to look more closely at the API to see if it doesn't make sense.

Either way the `create_parameter_service` like thing would be fine. However, I don't see how you get rid of the constructor that takes the various node interface pointers. I don't see any issue with that constructor -- we don't have to advise users to use it (i.e. use it our examples).

I also think we could just have a option for the node construction which causes the Node to create its own parameter service. I think that might be the "best" solution so long as the user doesn't need to do anything special with the parameter server's creation. This method could still use the "create_parameter_service` method in its implementation.

But if we have this we probably need a function which lets the user turn off the parameter service without destroying the node.

With that in mind, maybe it would be interesting to three modes for that option: "no parameter service", "parameter service on first parameter created, i.e. on demand" (could be the default), "parameter service explicitly on" (if you wanted external users to set parameters). That's a bit more complicated, but it buys us less overhead because simple nodes with no parameters would not default to having a parameter service, and at the same time nodes with parameters would not need to explicitly turn on the service. Nodes with parameters could also elect to explicitly turn off the parameter service. So I think it serves all the use cases as well as possible.

----

As for what we should do now, I'm in favor of:

- definitely have an option in the node options to have the node create a parameter service itself
- definitely have a function to create a parameter service and turn it off again (maybe let it go out of scope)
  - question: what happens if you call it multiple times? maybe it should just be something like `Node::enable_parameter_service()` and `Node::disable_parameter_service()`
- optionally have a more sophisticated pattern for controlling when and why the parameter service is enabled
- optionally have a way to check if a node has a parameter service externally
  - might just be look for a service to exist, but that has a race condition
  - might have a standard service on all nodes that list the node's features

Karsten1987, 2017-11-28:
> However, I don't see how you get rid of the constructor that takes the various node interface pointers.

I don't think there is, but we could provide two constructors as we do with the [ParameterClients](https://github.com/ros2/rclcpp/blob/022b2b1b807d8239b492e81a8fc156e36c60724d/rclcpp/include/rclcpp/parameter_client.hpp#L52-L64) currently. One which can take a node shared_ptr and then calls the second constructor with all single interfaces. With that the user could still use a relatively straightforward API, whereas the `create_parameter_service` uses the second constructor directly and thus prevents the weak_ptr exception.

> As for what we should do now, I'm in favor of:

I agree with your points and they make sense. I am not sure though, if we really should do that before the release or rather just focus on implementing the constructors  and create functions in order to let the users create parameter_clients and services inside their node constructors wo/ leading to exceptions.

dhood, 2017-11-28:
> However, I don't see how you get rid of the constructor that takes the various node interface pointers.

Yeah I meant the one that takes the Node shared pointer, not the one that takes the interfaces (parameter services only have the former at the moment (pt 1) but at the time I was using parameter clients as an example and they have both so it was ambiguous).

I've seen in todos that we have been meaning to make the parameter service start automatically with the node, and these seem like good proposals in that direction. As a more immediate step towards that we can add the `create_parameter_server` as proposed in pt 2 since we will need eventually to support remote parameter requests, continue starting it manually, and then consider how to start it automatically

wjwwood, 2017-11-28:
I think we should be careful in using the parameter client as the standard for the parameter service. They're used in fundamentally different ways. The client needs to be kept by the user so they can call methods on it. Whereas for the parameter server you just create it and forget it. So I think it's more appropriate for the node to create it and own it, especially since, unlike subscriptions, it doesn't take any user callbacks. The parameter related callbacks are given via node methods:

https://github.com/ros2/rclcpp/blob/7bbf5f6e5b225fcd05264ad72e5d19c290a652f0/rclcpp/include/rclcpp/node.hpp#L313-L321

If we're going to spend anytime on it now, I'd rather see the node do the work for us, but if you guys want to take the intermediate step to just have the function for manually creating it that's fine by me too.

dhood, 2017-11-28:
Correction: My point about "needing it anyway for remote parameter requests" only applies to parameters clients.

@wjwwood your last post has made me realise why the API is different for parameter services/clients: perhaps it's because, for parameter services at least, the API (and passing the node in) was only ever supposed to be temporary until it could be put behind the scenes. It didn't make sense to add `create_parameter_server` to the Node class if in the long run that would get started automatically :+1:  Then probably parameter clients just followed suit, though I still think parameter clients should be used via the node.

About parameter servers, you are right that it may not make sense to have the intermediate step, and given my "revelation" :wink: in the previous paragraph it probably just makes sense to use what's there for now for the paramter servers, making it clear that it's temporary, and then we can sort out the automatic starting later on.

So, circling back to my initial proposals, this is my take on it now:
1. Should we have a constructor for the ParameterService that takes the relevant interfaces, and the version that takes the node shared pointer will just forward to it (as is done for AsyncParametersClient)? This will allow users to create a parameter service in the constructor, but it will not be pretty. **We can leave the existing usage as is (it's temporary) and manually create the parameter server for each node after its construction. Users _won't_ be able to create parameter servers in the node constructor but this will not be the common usage pattern in the long-run anyway**.
1. Should we make convenience functions in the Node class for creating parameter services and parameter clients, so that the work of getting the node interfaces (shown above) is hidden from users? **Yes for parameter clients, because users should be able to easily create parameter clients in their Node subclass' constructor. Not needed for parameter servers because users won't create them manually.**

I still might have overlooked why the parameter _client_ usage is different to normal clients. Assuming we want parameter clients to look like normal clients, we want to add the `node->create_parameters_client` method. If there's actually a specific reason why they should have the node passed into the constructor we should go with the option of having the parameter client constructor that takes a `Node *` so that it can be called from within the node subclass' constructor.

wjwwood, 2017-11-28:
> Assuming we want parameter clients to look like normal clients, we want to add the `node->create_parameters_client` method.

I mean, that's fine, but I think that the parameters client is pretty different from the normal service clients (they are typed and are kept track of buy the node whereas the parameter clients are not templated/typed and the node doesn't need to keep track of it directly), so I don't think there is a need to make them the same though it's fine if they end up similar. The way I view the parameter client is that it is something that uses a Node, rather than something the node needs to keep track of or owns. Put another way, I'm not sure the Node needs to know about the existence of the parameter client. I don't suppose it matters, but I am worried about the number of methods we have on the node class.

Whether by a constructor which takes a `Node *` or by a `create_parameter_client` method on the Node class, it should be possible to create one in the constructor of a subclass. Though I'm not sure about using the parameter client in the constructor since it presumably needs to make service calls which will require the node being constructed to be put into an executor and spun.

closed 2017-12-04 by dhood