ros2/rclcpp #421: add test to check behavior when the queue size of the subscriber is reached

dirk-thomas, 2017-12-05:
To cover the issue #280.

Karsten1987, 2018-02-26:
@dirk-thomas I'll assign you to this ticket and given that the referencing issue is closed, I'll add this ticket to the bouncy milestone.
Feel free to re-assign or remove it from the milestone if you think so.

dirk-thomas, 2018-02-26:
Since I won't be working on this any time soon I will unassign myself. I also don't think having a test is critical for the next release either and will therefore also remove the milestone.

dirk-thomas, 2018-02-26:
I don't think `untargeted` is a good choice either. That will imply that we will "never" take another look at this ticket. Adding test coverage for this seems to be a reasonable thing in the future though.

closed 2026-10-14 by ahcorde