ros2/rclcpp #438: rclcpp logging still uses fprintf

guillaumeautran, 2018-02-16:
rclcpp logging still uses fprintf all over the place. We should align the logging with the standard RCLUTILS_LOG_XXX macros.

wjwwood, 2018-02-16:
Indeed that's a good goal, though I think @dhood might have been waiting to resolve some lingering logging pr's before doing this? Maybe not. Either way, help is appreciated.

dhood, 2018-02-16:
nah, I just added some logging, didn't convert all of the existing print statements

wjwwood, 2018-02-16:
Cool, then just something that needs to be done. I put the "help wanted" tag on it.

guillaumeautran, 2018-02-16:
I've got a patch almost ready to go (doing some testing). Will push it up soon.

dhood, 2018-04-27:
closed by https://github.com/ros2/rclcpp/pull/439

closed 2018-04-27 by dhood

mikaelarguedas, 2018-04-27:
What should we do for the commented out (f)printf calls still in the repo?
There seem to also be some printf still in use for the signal handlers.

dhood, 2018-04-27:
> What should we do for the commented out (f)printf calls still in the repo?

I imagine it will be updated when [that TODO](https://github.com/ros2/rclcpp/commit/49e9ed34b30202a98fd5aa825130e910dc813d34) gets addressed (unless you're speaking about something else)

> There seem to also be some printf still in use for the signal handlers.
Also has a [todo](https://github.com/ros2/rclcpp/blob/07e5be76218bd830c787b1d5c4af12e9eb6d5f6e/rclcpp/src/rclcpp/utilities.cpp#L138) that would need a decision (and it would need to update the behaviour on launch_testing if we swap) which I didn't think was included in the scope of this PR, but perhaps it was.

mikaelarguedas, 2018-04-27:
> I imagine it will be updated when that TODO gets addressed (unless you're speaking about something else)

It seems that we can easily miss it when we adress the TODO as the TODO doesnt mention it.
https://github.com/ros2/rclcpp/pull/466