ros2/rclcpp #441: Make number of threads in MultiThreadedExecutor configurable

matt-attack, 2018-02-20:
Currently the MultiThreadedExecutor just uses the number of hardware cores/threads as the number of threads to use: https://github.com/ros2/rclcpp/blob/master/rclcpp/src/rclcpp/executors/multi_threaded_executor.cpp#L29

It would be useful for it to be optionally configurable through an argument to the constructor so that one can effectively limit the maximum resource usage, much like how this could be done with nodelets in the past.

dirk-thomas, 2018-02-20:
Please consider to provide a PR for this, e.g. exposing the limit as an optional constructor argument.

jlblancoc, 2018-02-20:
Why not adding a new field to ExecutorArgs instead of adding a ctor parameter?

dirk-thomas, 2018-02-20:
> Why not adding a new field to ExecutorArgs instead of adding a ctor parameter?

That would be an option too. (It might not apply to all executors though, e.g. single threaded ones.)

PR #442 (2018-03-30): Add argument for thread count to multithreaded executor
Resolves #441
closed 2018-03-30 by tfoote