ros2/rclcpp #455: Client (async_send_request) could not return when the data (message) is lost

gong0029, 2018-03-28:
In ROS2 client, if the message is lost, the client function "async_send_request" will not return any more. I try to solve the problem in rmw level and hack the code in function "rmw_take_response" to return RMW_RET_ERROR if the message is lost, but it doesn't work.

The reason is that: 

In function "Executor::execute_client":

   if (status == RCL_RET_OK) {
        client->handle_response(request_header, response);
    } else if (status != RCL_RET_CLIENT_TAKE_FAILED) {
       fprintf(stderr,
      "[rclcpp::error] take response failed for client of service '%s': %s\n",
       client->get_service_name().c_str(), rcl_get_error_string_safe());
      rcl_reset_error();
  }

 If the return value is Error, the function "handle_response" could not be executed.
 
However, in function  "handle_response"
{
    ...
    call_promise->set_value(typed_response);
    callback(future);
}

If the  "callback(future)" is not executed, the "async_send_request" will not return f.

So I wonder if this problem can be solved in rclcpp? Because in our implementation in rmw, the message may be lost.

wjwwood, 2018-03-29:
I don't understand your question, because `async_send_request` should not block depending on whether the request message is sent or received successfully. It should send (or queue to send) and then return immediately.

Also, `async_send_request` should not directly be calling `rmw_take_response` and therefore should not block it from returning. The executor will call that if and only if `rmw_wait` wakes up when it was waiting on the `rmw_client_t` handle and it was not set to null (similar fashion to how `select()` works).

Similarly, `rmw_take_response` should never block because it should take data if available but if not available it should return `RMW_RET_OK` but the `taken` bool it uses as an "output" variable should be `false`:

https://github.com/ros2/rmw/blob/master/rmw/include/rmw/rmw.h#L232

For `rcl`, the function should return `RCL_RET_CLIENT_TAKE_FAILED`:

https://github.com/ros2/rcl/blob/master/rcl/include/rcl/client.h#L281

gong0029, 2018-03-29:
I mean async_send_request will not return until the response is received successfully (because async_send_request need to return the response value).

If you set *taken=false in "rmw_take_response", then "rcl_take_response" will return  RCL_RET_CLIENT_TAKE_FAILED and "execute_client" will not execute "handle_response". Then "async_send_request" will not return.

wjwwood, 2018-03-29:
> I mean async_send_request will not return until the response is received successfully (because async_send_request need to return the response value).

I don't believe that's true, as you can see here that function does not call `rcl_take_response`:

https://github.com/ros2/rclcpp/blob/ef17ec624849d269429166fc4b7dca4d93d15895/rclcpp/include/rclcpp/client.hpp#L223-L247

And even if it did try to take a response, it would just fail immediately since you didn't wait in between.

Perhaps you could post some steps to reproduce it or at least give some example code which you expect to work one way but does something different.

gong0029, 2018-03-29:
In "async_send_request", the function will be blocked at "return f;"(line 256) until the function "handle_response" execute "callback(future);"(the last line). But if you set *taken=false in "rmw_take_response", then "execute_client" will not execute the function "handle_response".

If you wangt to reproduce it, you can simply change the file rmw_respomse.cpp (for example we use fastrtps) and add " *taken=false; " before "return RMW_RET_OK;" in rmw_take_response. 

In this file:
    https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_cpp/src/rmw_response.cpp
line 66

And then run the "demo_node_cpp/add_two_ints_client"  and "demo_node_cpp/add_two_ints_server".

You can see the client will not return.

wjwwood, 2018-03-31:
> In "async_send_request", the function will be blocked at "return f;"

Hmm, I don't think that's the way futures are supposed to work. My expectation is that `return f;` will simply be copying a shared pointer to a `std::future` (which is what `SharedFuture` is), only when you try to wait on the future would it block.

But I will try to reproduce your example and see if I can reproduce it.

gong0029, 2018-04-02:
Fine. Thanks for your help. I don't  exactly know how "std::future" works. I wonder if "return f" does not block the function "async_send_request", how to guarantee to obtain the response from the RPC call?

gong0029, 2018-04-16:
Did you successfully reproduce the example? @wjwwood

wjwwood, 2018-04-16:
@gong0029 I haven't had time to reproduce it. Hopefully, I'll get to it soon.