ros2/rclcpp #457: Test of node->now() flaky on OS X

dhood, 2018-03-30:
[This test](https://github.com/ros2/rclcpp/blob/787de6ebf1a8748cd36802f8c1e6a41fc0f85fed/rclcpp/test/test_node.cpp#L113) is flaky:
```
TEST_F(TestNode, now) {
  auto node = std::make_shared<rclcpp::Node>("my_node", "/ns");
  auto clock = node->get_clock();
  auto now_builtin = node->now().nanoseconds();
  auto now_external = clock->now().nanoseconds();
  EXPECT_GE(now_external, now_builtin);
  EXPECT_LT(now_external - now_builtin, 50000L);
}
```

The failures that I've seen are always on OS X, at the last line.

Here are some examples:
https://ci.ros2.org/view/nightly/job/nightly_osx_repeated/1039/testReport/junit/rclcpp/TestNode/now/ (mini1)
https://ci.ros2.org/view/nightly/job/nightly_osx_repeated/1046/testReport/junit/rclcpp/TestNode/now/ (mini2)
https://ci.ros2.org/view/nightly/job/nightly_osx_repeated/1047/testReport/junit/rclcpp/TestNode/now/ (mini3)

The time difference is ~1ms (0.98, 1.33, 1.27, ...) instead of the 0.05ms being compared against.

@tfoote given the implementation details, does anything come to mind about what could cause the >1ms difference? Or could this could be reasonable behaviour?

tfoote, 2018-03-30:
That's very surprising. This test was setup as a tripwire and should not be anywhere near performance limits. Otherwise we can't do timing in under ~1ms.

The tolerance is there to account for calling "now" twice in a row via the two different APIs.  `node->now()` and  [clock `now()` ](https://github.com/ros2/rclcpp/blob/787de6ebf1a8748cd36802f8c1e6a41fc0f85fed/rclcpp/src/rclcpp/clock.cpp#L76) Which node->now() [just passes through to](https://github.com/ros2/rclcpp/blob/787de6ebf1a8748cd36802f8c1e6a41fc0f85fed/rclcpp/src/rclcpp/node.cpp#L216) 

Which just passes through to [`rcl_clock_get_now`](https://github.com/ros2/rcl/blob/master/rcl/src/rcl/time.c#L229) which ends up with [`rcutils_system_time_now`](https://github.com/ros2/rcutils/blob/master/src/time_unix.c#L49) Which does have special behavior for OSX.

dirk-thomas, 2018-03-30:
As far as I understand the [underlying code](https://github.com/ros2/rclcpp/blob/45dcd0c6e58bf54d771ae8635d30cc91f8a08e2d/rclcpp/src/rclcpp/node.cpp#L207-L217) both ways to retrieve the current time are identical.

Therefore `EXPECT_GE` basically checks that the returned time is not going backwards. That makes sense even though a single comparison might not say a lot.

But what is the goal of `EXPECT_LT` here? Can someone describe what this is supposed to be testing? The scheduler might decide anytime to take away control of the CPU from this process in order to give it to a different process. If that happens between the two `now()` calls there is an arbitrary time between these time stamps. Based on that the check doesn't make much sense.

tfoote, 2018-03-30:
The goal of this check is to make sure that it's still using the same clock and not somehow completely different. There's some threshold that should be reasonable to expect it to be closer than.

A 1ms hiccup between two consecutive calls in the same thread without a yield seems to be quite large and possibly indicative of another issue.

dirk-thomas, 2018-03-31:
> A repeatable 1ms hiccup between two consecutive calls in the same thread without a yield seems to be quite large and possibly indicative of another issue.

Why do you think so? As as describe above you might just "loose" the CPU and on a system with a non-realtime kernel it might take hundreds of milliseconds until the code path continues.

dhood, 2018-04-02:
> As as describe above you might just "lose" the CPU and on a system with a non-realtime kernel it might take hundreds of milliseconds until the code path continues. 

Sounds like this underpins a number of other test failures we see sometimes, e.g.
These `rcl` timer failures on Windows (e.g timer shouldn't be ready but it is, timer took too long to wake up): https://ci.ros2.org/view/nightly/job/nightly_win_rep/1034/testReport/rcl/
Or this `rclpy` timer failure on OS X (waits for 0.0005s for a timer with period 0.001s, checking it doesn't receive anything, but it does. Underneath it must have taken longer than expected to wake up): https://ci.ros2.org/view/nightly/job/nightly_osx_repeated/769/testReport/test/test_timer/test_timer_zero_callbacks1000hertz/

What is the appropriate response from our side, knowing that we can't guarantee that these tests pass the way that they're written? They still provide useful info that "most of the time" performance is within the expected tolerance, so if we disable them completely we'll lose that info.
Do we continue to keep these tests around under the understanding that they will occasionally fail, and only get concerned if they start failing very often?

dirk-thomas, 2018-04-02:
Imo unit tests should not test for performance explicitly. They should be satisfyable even on "slow" system or systems with concurrent load.

Separate performance tests could ensure that we are satisfied with the latency (whatever else) but those shouldn't cause flaky failures on our regular builds.

dhood, 2018-04-02:
I suppose the approach will depend on the "type" of test then.

E.g. "timer shouldn't be ready so soon, but for some reason it is" is difficult to _ever_ guarantee on slow systems the way that the test is written, so could be removed/disabled.

While "timer is expected to wake up eventually but took too long to wake up" is _possible_ to satisfy and might just need its tolerance relaxed. That's assuming it's the only way we are testing that timers that are supposed to wake up do wake up; if it's a separate performance test, disable it.

mikaelarguedas, 2018-04-05:
Conclusion from triage meeting:
Unit test should not test performance or have very relaxed acceptance criterias to account for the (non) real-time nature of the platform they run on.
We can keep the low frequency tests as regular unit tests.
The performance ones should be adapted to have more permissive acceptance criteria and labeled as performance tests.
example of tests to be adapted (not a full list just those that come to mind):
- rclpy timer tests: https://github.com/ros2/rclpy/blob/master/rclpy/test/test_timer.py
- rcl timer tests: https://github.com/ros2/rcl/blob/master/rcl/test/rcl/test_timer.cpp
- rclcpp ROS Time tests

closed 2018-08-17 by dhood