ros2/rclcpp #462: Ability to serialize messages without publishing

jespersmith, 2018-04-16:
## Feature request

#### Feature description
It can be advantageous to send the messages over a custom protocol (to allow sending data over WAN or non-network data link like SPI etc) while still leveraging the serialization that ROS2 provides. 

A function to the RCLCPP to serialize ROS2 messages to a byte array (maybe std::vector<uint8_t>) would help to implement this functionality.

#### Implementation considerations

Different middle-wares could result in different serialization output. Ideally they are compatible.

wjwwood, 2018-04-16:
This is related to the set of changes connected to this pr: https://github.com/ros2/rclcpp/pull/388

That lets you publish and take messages in the serialized format (`uint8_t` buffer). We'll need this feature you've described (as well as its corollary) to properly implement rosbag's as well.

/cc @Karsten1987

Karsten1987, 2018-04-16:
The basic outline for the new API can be found in the demo PR https://github.com/ros2/demos/pull/185  . 
I am currently working on this, still having some code cleanup and memory management organization to do on it. That is to say, that it's not ready to merge yet.
There is currently only a publish/take function to publish and receive messages in a serialized form. A second set of function is yet to come which allows you to serialize messages without having them sent.

wjwwood, 2020-03-19:
@Karsten1987 what's the status on this? Is it still not possible to manually serialize/deserialize a message?

Karsten1987, 2020-03-19:
Thanks for the reminder. I am going to close this as I consider the requested feature is implemented. 

The RMW interface has functions to (de-)serialize ROS messages into a `uint8`-array and vice versa:
https://github.com/ros2/rmw/blob/master/rmw/include/rmw/rmw.h#L516-L559

As already stated, different RMW implementations might differ in which format they use to serialize. In order to get that information, we have a separate function in RMW to get this identifier. All supported DDS implementations should be compatible to each other as all serialize via CDR:
https://github.com/ros2/rmw/blob/master/rmw/include/rmw/rmw.h#L109-L124

@jespersmith please feel free to further comment on this ticket (or re-open it) in case your concerns are not addressed.

closed 2020-03-19 by Karsten1987