ros2/rclcpp #475: Expected vs Unexpected parameters Node::create_param()

sloretz, 2018-05-18:
## Feature request
Allow node author to define expected parameters and what happens when an unexpected parameter is set.

#### Feature description

This issue is a first step towards larger features like parameter constraints (allowed types, read-only parameters, allowed ranges, etc...). These will be ticketed separately. 

**Use case 1:**
A node author knows all parameters at compile time. Unexpected parameters at run time should be rejected. For example, there may be a typo in a parameter name being set at run time.

**Use case 2:**
A node author may not know all possible parameters at compile time. For example, `costmap_2d::ObstacleLayer` does not know all of the parameter names until after `observation_sources` is set. Unexpected parameters should be allowed so they can be used later when a node has more information.

#### Implementation considerations

Some parameters functionality will move to rcl (ros2/rcl#194). I'm assuming the client library (`rclcpp`, `rclpy`, etc) will be responsible for parameter storage including default values and having knowledge of expected versus unexpected parameters.

Defining a parameter

```c++
/// Create a parameter with an optional initial value
/*
 * \param[in] name a '.' delimited parameter name
 * \param[in] initial_value the initial value of the parameter, or omit if the value should be uninitialized.
 * \raise an exception if an error occurs: parameter with same name already exists, or parameter name is invalid
 */
void
NodeParameters::create_parameter(const std::string & name, ParameterVariant initial_value = ParameterVariant());
```

At a minimum a node author should be able to set a `bool` indicating if unexpected parameters are allowed. Calling a user callback with unexpected parameters would allow a user to do more complicated logic.

sloretz, 2018-05-21:
Related to the discussion concluded here: https://github.com/ros2/rclcpp/pull/309#issuecomment-286906648

clalancette, 2024-01-19:
> Allow node author to define expected parameters and what happens when an unexpected parameter is set.

With the design of parameters to be declared, I think that this functionality is basically already present.  I'm going to close this out, but please feel free to reopen if you disagree.

closed 2024-01-19 by clalancette