ros2/rclcpp #514: Nodes default to using ROS time if /clock is published

dhood, 2018-07-17:
The presence of a publisher on the `/clock` topic causes nodes with unset `use_sim_time` parameter value to start using sim time. [See this test](https://github.com/ros2/rclcpp/blob/d298fa4445c49ce2229752f79e92533bab1caaaa/rclcpp/test/test_time_source.cpp#L112..L113).

I haven't found a discussion where we decided that should be the case, so wanted to highlight the current state of ROS time being "opt out" instead of "opt in" and verify it's what we want.

tfoote, 2018-07-25:
This was the original behavior in ROS. But it looks like it was updated circa 2010 to require the parameter to be set before startup of the node otherwise it won't subscribe to `/clock` instead of that just being best practice. We should probably switch to the existing behavior where the parameter is required to subscribe. Though I might suggest that we not require it prior to launch since we can now get a callback on a parameter change. And subscribe if the parameter is set to true.

closed 2018-09-25 by dhood

dhood, 2018-09-25:
the use_sim_time solely is what determines if ROS time is active in #559. /clock subscription is only created if it is set to true. Parameter callbacks are used, so this subscription will be created if the parameter is set at any time during the lifetime of the node, and removed if the parameter is set to false.

https://github.com/ros2/rclpy/issues/240 tracks doing something similar in rclpy

PR #559 (2018-09-25): Don't auto-activate ROS time if clock topic is being published
closes #514 and #515

Previously, the clock subscription was always created. When messages were received on the clock topic, ROS time was set to active (#514), even if `use_sim_time` was set to false (#515).

Now, ROS time will only be set active if the `use_sim_time` parameter is explicitly set true. The subscription is only created if the parameter is set (can be set after node startup).

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

In progress because CI might uncover things relying on the previous  [... cut]