ros2/rclcpp #515: Node unable to avoid using sim time if /clock published

dhood, 2018-07-17:
Even if a node has its `use_sim_time` parameter set to false, if a message is received by the TimeSource on the `/clock` topic, the node's clock will start using sim time.

This branch has added a test that fails but I would expect to pass: https://github.com/ros2/rclcpp/blob/07567dfbcd56803d0cb5ff8afe23e6b3171d02b6/rclcpp/test/test_time_source.cpp#L339..L341

The presence of a publisher on the clock topic currently causes nodes with unset `use_sim_time` parameter value to start using sim time (see https://github.com/ros2/rclcpp/issues/514 if you want to discuss that), but I don't think it should happen for those that have explicitly set the parameter to false.

If others agree we can label this as a bug and then only enable ROS time [upon receipt of /clock msg](https://github.com/ros2/rclcpp/blob/ae6f8e3e9a6cab18877ce885ad795c1c339b7d8e/rclcpp/src/rclcpp/time_source.cpp#L203) if the parameter has not explicitly been set. That would keep the rest of the existing behaviour the same (again, if you want to discuss the behaviour on unset `use_sim_time` parameter, see https://github.com/ros2/rclcpp/issues/514).

wjwwood, 2018-07-17:
In my opinion, it should definitely be opt-in (presence of a publisher on `/clock` is not sufficient), though maybe a one time warning is appropriate if `use_sim_time` is unset but there is a publisher (it might be an indication that there is a configuration issue). Explicitly setting `use_sim_time` to False could disable the warning.

sloretz, 2018-07-17:
+1 to opt-in

Is a `/clock` subscription necessary when `use_sim_time` is `false`? Not having it would prevent a warning like @wjwwood described. It also would add a delay if the subscription had to be created later when `use_sim_time` is dynamically enabled, but I'm not sure if that's a problem.

tfoote, 2018-07-25:
The behavior in ROS1 is to not even subscribe unless the parameter is set. It's actually required that the parameteter is set before startup as the subscription is only created during the startup phase of a ROS1 node. I implemented this off of memory from the first implementation of ROS1 time but that evolved since that generation. As I mentioned in #514 we might want to take advantage of the parameter callbacks and support switching based on the parameter instead of requiring it at startup. Code can be robust to the change in time by using the TimeJump callbacks.

dhood, 2018-09-25:
#559 made it so time source's use of sim time only depends on the use_sim_time parameter, not the presence of a clock publisher. [This test](https://github.com/ros2/rclcpp/pull/559/files#diff-8b3e7c8b544d7848db66291e08959f94R365) was added to check that sim time doesn't activate even if a clock publisher is present if the `use_sim_time` parameter is set to false.

closed 2018-09-25 by dhood