ros2/rclcpp #518: Add way to check if init has been called (i.e. isInitialized)

chapulina, 2018-07-19:
## Feature request

#### Feature description

It would be convenient to have a way to check if `rclcpp::init` has been called, similar to [ros::isInitialized](http://docs.ros.org/diamondback/api/roscpp/html/namespaceros.html#accd0b48f347e82ae96b261f680af41dd).

#### Implementation considerations

If this has been left out due to some design decision, I suppose users can live without it, and instead try to catch specific errors resulting from not calling init. I searched a bit and couldn't find an explanation, so I imagine it may just be in the backlog?

Thanks!

sloretz, 2018-07-19:
I was going to say `rclcpp::ok()`, but it looks like it returns `true` if `rclcpp` hasn't been initialized. Maybe it should return `false` instead?

mikaelarguedas, 2018-07-19:
would [`rclcpp::ok()`](https://github.com/ros2/rclcpp/blob/8f2052d65a8b05e19958db5484f5425c688098b8/rclcpp/include/rclcpp/utilities.hpp#L86-L88) work for your use case? (though there may be some subtle differences as it's using a global that I think would default to returning true). We could make rclcpp::init assign a value to the global and flip the default

There is indeed a related issue in the backlog: https://github.com/ros2/rclcpp/issues/3

chapulina, 2018-07-19:
Yeah if `g_signal_status` is [initialized](https://github.com/ros2/rclcpp/blob/master/rclcpp/src/rclcpp/utilities.cpp#L43) to something other than 0 and set to 0 later, then maybe `rclcpp::ok` could be enough.

sloretz, 2018-07-19:
It  seems like the discussion in the waffle meeting today leaned towards having separate APIs for tracking whether a signal has been received versus whether rclcpp is currently initialized.

wjwwood, 2018-07-26:
Closed by #522.

closed 2018-07-26 by wjwwood