ros2/rclcpp #521: add option to disable all ROS signal handling

wjwwood, 2018-07-19:
Right now we unconditionally install our own signal handler:

https://github.com/ros2/rclcpp/blob/8f2052d65a8b05e19958db5484f5425c688098b8/rclcpp/src/rclcpp/utilities.cpp#L174-L193

We should have an option to avoid this if the user desires, and along with that any documentation needed to describe what they need to do themselves since the signal handler is no longer doing it.

wjwwood, 2020-03-23:
This is no longer the case with https://github.com/ros2/rclcpp/pull/587 and related prs.

closed 2020-03-23 by wjwwood

EricCousineau-TRI, 2023-11-10:
For those coming to this issue looking for an answer, here are the relevant API docs for Humble for `rclcpp::init(..., SignalHandlerOptions)`:
http://docs.ros.org/en/humble/p/rclcpp/generated/function_namespacerclcpp_1a026b2ac505c383735117de5d1679ed80.html
http://docs.ros.org/en/humble/p/rclcpp/generated/enum_namespacerclcpp_1a14a5d1b5fa11030e4141fd79e36cf2fd.html#_CPPv4N6rclcpp20SignalHandlerOptionsE