ros2/rclcpp #529: Allow `Parameter` API to get/set maps of values

clalancette, 2018-08-06:
## Feature request

#### Feature description

There are some users of the Parameter API in ROS1 that depend on getting a map of values, e.g. https://github.com/ros-teleop/teleop_twist_joy/blob/indigo-devel/src/teleop_twist_joy.cpp#L78 .  It would be great to have a helper API to do this in ROS2 as well, to reduce the amount of boilerplate that users would have to do to get parameters one at a time.

mikaelarguedas, 2018-08-06:
Permalink: https://github.com/ros-teleop/teleop_twist_joy/blob/10d79bebbf112f95f32713895f7cd0668faa83fc/src/teleop_twist_joy.cpp#L78

dirk-thomas, 2018-08-06:
> ... to reduce the amount of boilerplate that users would have to do to get parameters one at a time.

I am not sure what you mean with "one at a time".

Couldn't you create a "syntactic sugar" API which wraps the existing [get_parameters](https://github.com/ros2/rclcpp/blob/4ddb76f466d331e4ca8ff4df7ba25a5dce8496af/rclcpp/include/rclcpp/parameter_client.hpp#L188-L190 method and instead of returning a `std::vector<rclcpp::Parameter>` return a `std::map<std::string, rclcpp::ParameterValue>`?

nuclearsandwich, 2018-09-07:
> Couldn't you create a "syntactic sugar" API which wraps the existing get_parameters method and instead of returning a std::vector<rclcpp::Parameter> return a std::map<std::string, rclcpp::ParameterValue>?

From talking with @clalancette about this a get_parameters overload which can return a map or fill a provided one would be a good enhancement, although not needed for the work on teleop_twist_joy.
Another method that would be useful (but again isn't necessary for teleop_twist_joy) is one which can get an entire prefix. An example would be something like `std::map<std::string, rclcpp::ParameterValue> get_parameters_by_prefix(std::string prefix);` Given the parameters in our [example on the wiki](https://github.com/ros2/ros2/wiki/Node-arguments#parameters) this method would be used to get a map of all parameters under the prefix `some_lists`