ros2/rclcpp #531: add "NodeHandle" like concept

wjwwood, 2018-08-10:
In ROS 1, there's a construct in `roscpp` called a `NodeHandle`, and it has most of the interface of the Node singleton, but can additionally have "extra" namespace. Having a node handle instance with a namespace does not affect the node namespace/name combo, but all things created using the namespace node handle will have the extra namespace added to it, e.g. publishers, subscriptions, services, parameters, etc...

From https://wiki.ros.org/roscpp/Overview/NodeHandles:

> The ros::NodeHandle class serves two purposes. First, it provides RAII-style startup and shutdown of the internal node inside a roscpp program. Second, it provides an extra layer of namespace resolution that can make writing subcomponents easier.

In `rclcpp`, the `Node` class provides the RAII style lifecycle of the node, so this issue is just about having a new concept in `rclcpp` that satisfies the "provides an extra layer of namespace resolution that can make writing subcomponents easier" part.

However, there is a question about lifetime of the node handle: should it be up to the user to ensure that the node instance out lives all node handles, or should node handles have a weak reference to the node?

Unlike the constructible `ros::NodeHandle`, this new object would be generated by a factor function of the original Node. e.g. `node->get_node_handle("namespace", remapping args, additional parameters, etc.)`.

Again, like the node handles in `roscpp`, it should be possible to build a hierarchy of these objects, something like this:

```c++
auto node_ptr = std::make_shared<rclcpp::Node>("my_node", "my_ns");
auto sub_node = node_ptr->create_sub_node("foo_ns");
auto another_sub_node = sub_node->create_sub_node("bar_ns");
// topic would be '/my_ns/foo_ns/bar_ns/chatter'
auto sub = another_sub_node->create_subscription<...>("chatter", ...);
```

This request is being driven by existing use cases in the porting of gazebo ros packages, and is especially useful if you have components which may be "attached" to nodes in arbitrary configurations. These components would take something that looks like a node and use it to create pub/sub, services, etc... but ideally would not need to manually apply namespaces itself.

----

Summary of design issues and goals:

- Provide API of node, but modify namespace of topics, services, parameters, etc...
- Provide custom but scoped remappings
- Provide additional parameter value initialization/declaration (see: https://github.com/ros2/rclcpp/blob/6b34bcc94c12e2213b6de7af57c3d10109402906/rclcpp/include/rclcpp/node.hpp#L101)
- Nestable
- Name bikesheding, how to call it?
  - considerations: it's not a Node, how to indicate? can provide extra namespace, but also other things
  - ideas: `NodeHandle`, `NodeProxy`, `SubNode`, `NodeView`, `NodeNamespace`
- How to handle lifecycle of handle with respect to lifecycle of actual node?
- How to avoid extensive boilerplate required to implement our composition design (Node, "NodeHandle", LifeCycleNode, etc...)?

wjwwood, 2018-08-11:
I'll add, after a long discussion with @dhood and @tfoote, that an acceptable and perhaps preferrable solution to this is that the `rclcpp::Node` class becomes one of these "node handle facades", and the first one you get simply has no _additional_ namespacing, remappings, or initial parameter values. That way there's not two API's, one for "full nodes" and one for "partial node handles". This is also how it works in ROS 1 (you always get a `NodeHandle` object in ROS 1).

In this case the new "concept" described here would also control the lifecycle of the node, since all instances of the facade would have a shared ownership of the "node implementation" which actually does the work. So similar to ROS 1, a node would not disappear from the graph until all of the node facades have gone out of scope.

fmrico, 2018-11-02:
Hi,

I just open a PR for this in #578 

Hope it helps

sloretz, 2018-11-09:
In ROS 1 trying to use a private name when creating a publisher or subscriber fails.

ROS 1
```c++ 
ros::NodeHandle n;
ros::Publisher pub = n.advertise<std_msgs::String>("~/n/private", 1000);
/*
terminate called after throwing an instance of 'ros::InvalidNameException'
  what():  Using ~ names with NodeHandle methods is not allowed.  If you want to use private names with the NodeHandle interface, construct a NodeHandle using a private name as its namespace.  e.g. ros::NodeHandle nh("~");  nh.getParam("my_private_name"); (name = [~/n/private])
*/
```

 However, ROS 2 APIs like `create_publisher()` allow private names to be passed in. If the ros2 `Node::create_...()` functions are moved to sub nodes, then private and fully qualified topic names passed to them need to be handled. The table below proposes how to handle all the cases I can think of. Thoughts?

Assuming a user creates a `Node` with namespace `/orig_ns` and name `nodename` followed by a `SubNode`.

| Sub node namespace | Input topic name | Proposed output |
| --- | --- | --- |
| `rel_ex` | `rel_on` | `/orig_ns/rel_ex/rel_on` |
| `rel_ex` | `~/priv_on` | `/orig_ns/rel_ex/nodename/priv_on` |
| `rel_ex` | `/fqn_on` | `/fqn_on` |
| `~/priv_ex` | `rel_on` | `/orig_ns/nodename/priv_ex/rel_on` |
| `~/priv_ex` | `~/priv_on` | `/orig_ns/nodename/priv_ex/nodename/priv_on` |
| `~/priv_ex` | `/fqn_on` | `/fqn_on` |
| `/fqn_ex` | `rel_on` | `/fqn_ex/rel_on` |
| `/fqn_ex` | `~/priv_on` | `/fqn_ex/nodename/priv_on` |
| `/fqn_ex` | `/fqn_on` | `/fqn_on` |

wjwwood, 2018-11-09:
Since `/~` and `foo/~/bar` are not a valid names (see: https://design.ros2.org/articles/topic_and_service_names.html#ros-2-name-examples), I kind of think of `~` as implicitly absolute. So I don't know if I would treat it as a relative name w.r.t. a "sub node". Instead, I would have expected it to be treated like an absolute name.

I'm not 100% certain, but that's my gut feeling.

Maybe it's not a bad idea to enforce a similar rule as ROS 1. We didn't until now, but on the other hand we didn't have the "node handle" or "sub node" concept either.

Other than that, your table makes sense to me.

wjwwood, 2019-02-21:
Closed with #581

closed 2019-02-21 by wjwwood