ros2/rclcpp #532: Remove asymmetry in subscribing/publishing serialized messages

Martin-Idel-SI, 2018-08-16:
## Small Feature request

Remove asymmetry in subscribing/publishing to serialized message

#### Current behavior

I can subscribe to a serialized message directly by calling `create_subscription` with a callback of the form
```
[](std::shared_ptr<rcutils_char_array_t> msg) { ... }
```
However, I call publish handing in a raw pointer:
https://github.com/ros2/rclcpp/blob/6b34bcc94c12e2213b6de7af57c3d10109402906/rclcpp/include/rclcpp/publisher.hpp#L280

#### Improved behavior

I can call publish also by handing in a `shared_ptr` of a serialized message.

sloretz, 2018-08-16:
ping @Karsten1987 , mind having a look at this one?

Karsten1987, 2018-08-24:
@Martin-Idel-SI Sorry for the late response. I somehow didn't see that issue in my inbox.
I'll have a look at it.

closed 2018-08-24 by Karsten1987