ros2/rclcpp #539: add_two_ints_client Ctrl-C fail

reed-lau, 2018-08-21:
fastrtps ,opensplice 

change add_two_int_client.cpp to the loop version

it could not be interrupted by Ctrl-C
and will output  [ERROR] [add_two_ints_client]: Interrupted while waiting for response. Exiting.

mikaelarguedas, 2018-08-21:
Thanks @reed-lau for reporting. Unfortunately this is not enough information, can you please fill out the entire issue template including your platform, ROS 2 version, exact steps to reproduce the issue you're facing etc.

Thanks

reed-lau, 2018-08-22:
## Bug report

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - from source
- Version or commit hash:
  - 5fb14ddefe67f0ce0086f67d66c6ed0204635695 0.5.1 bouncy
- DDS implementation:
  - FastRtps and Opensplice
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
1.  Change the add_two_ints_client.cpp to a loop version

diff --git a/demo_nodes_cpp/src/services/add_two_ints_client.cpp b/demo_nodes_cpp/src/services/add_two_ints_client.cpp
index f92670b..31e8287 100644
--- a/demo_nodes_cpp/src/services/add_two_ints_client.cpp
+++ b/demo_nodes_cpp/src/services/add_two_ints_client.cpp
@@ -86,11 +86,16 @@ int main(int argc, char ** argv)
 
   // TODO(wjwwood): make it like `client->send_request(node, request)->sum`
   // TODO(wjwwood): consider error condition
-  auto result = send_request(node, client, request);
-  if (result) {
-    RCLCPP_INFO(node->get_logger(), "Result of add_two_ints: %zd", result->sum)
-  } else {
-    RCLCPP_ERROR(node->get_logger(), "Interrupted while waiting for response. Exiting.")
+  int i =0;
+  while (1)
+  {
+      request->b = i++;
+      auto result = send_request(node, client, request);
+      if (result) {
+          RCLCPP_INFO(node->get_logger(), "Result of add_two_ints: %zd", result->sum)
+      } else {
+          RCLCPP_ERROR(node->get_logger(), "Interrupted while waiting for response. Exiting.")
+      }
   }
 
   rclcpp::shutdown();

2. compile it using colcon build --packages-select demo_nodes_cpp -cmake-args -DCMAKE_BUILD_TYPE=Debug

3.  run the add_two_ints_server (in demo_nodes_cpp package)

4. run the add_two_ints_client ( in demo_nodes_cpp package), after it can output the result Ctrl-C it

then the console will output
[ERROR] [add_two_ints_client]: Interrupted while waiting for response. Exiting. for ever and Ctrl-C could not stop it any more.

#### Expected behavior
stop the add_two_ints_client process
#### Actual behavior
could not stop
#### Additional information
do I have to handle the case by myself. add a signal handler ? 
## Feature request
Ctrl-C could set a global state to stop the ros2-system's wait_set instead of one time. or add one api to exit the loop.

dirk-thomas, 2018-08-27:
Your code uses a `while (1)` loop so the signal handling in ROS won't be able to "escape" that one. Please try using `while (rclcpp::ok())` instead.

If this change address your problem please go ahead and close this ticket.

reed-lau, 2018-08-28:
yes it worked

closed 2018-08-28 by reed-lau