ros2/rclcpp #544: Allow subscribing without template parameter

Martin-Idel-SI, 2018-08-28:
## Feature request

#### Feature description
Currently, we can only subscribe to a topic via `node->create_subscription<MessageT>(<topic_name>, <callback>)` or similar calls. This requires the topic type to be known at compile time. 

It would be great to have a way to create a subscription without this template, passing type support into the subscription creation manually. 

*Use Case:* This would help us in `rosbag2`, where we need to subscribe to all available topics which we obviously don't know at compile time.

wjwwood, 2018-09-06:
Why isn't the template type `rmw_serialized_message_t` or a similar type?

wjwwood, 2018-09-06:
Or asked a different way, what would the signature of the callback for the subscription look like, currently it is something like:

https://github.com/ros2/rclcpp/blob/198c6daf49baf898ae0a91b7f8e6f90b4d513b86/rclcpp/include/rclcpp/any_subscription_callback.hpp#L45-L46

Martin-Idel-SI, 2018-09-10:
The signature of the callback is fine - and you can already subscribe to a topic using a callback with `const std::shared_ptr<rmw_serialized_message_t>` argument, i.e. you can do something like

```
node->create_subscription<std_msgs::msg::String>(topic_name, [](const std::shared_ptr<rmw_serialized_message_t> message () {...});
```

However, if you try to create a subscription directly to `rmw_serialized_message_t` (i.e. `create_subscription<rmw_serialized_message_t>`, you'll get a link error because typesupport is not available. If there was a template specialisation where the type would have to be added manually, I'd be happy.

Karsten1987, 2021-04-03:
After quite some time, I am happy to close this ticket with https://github.com/ros2/rclcpp/pull/1452

closed 2021-04-03 by Karsten1987