ros2/rclcpp #552: rclcpp_lifecycle: issue a warning if publishing on a not active publisher

Myzhar, 2018-09-06:
## Feature request

#### Feature description
Since a publisher can be active or not is should be useful for debug purpose to issue a WARN when publishing on a not active publisher

#### Implementation considerations
Add a RCLCPP_WARN at Line 78:
https://github.com/ros2/rclcpp/blob/4d67a8671b55b28a6ae63697310a11632a888ed4/rclcpp_lifecycle/include/rclcpp_lifecycle/lifecycle_publisher.hpp#L78

Karsten1987, 2018-09-06:
I believe the warning makes sense. Would you consider opening a PR for this?

Myzhar, 2018-09-27:
I'm adding the warning message, but I cannot understand how to do it.
Normally I use `RCLCPP_WARN`, but I'm not inside a node, so it cannot be called, right?

mikaelarguedas, 2018-09-27:
To use rclcpp logging macros you do need a `rclcpp::Logger`, the [demos repository](https://github.com/ros2/demos) contains some examples of passing a [node logger to a function](https://github.com/ros2/demos/blob/e3b99c4cfdbbb50455d4a1fab17324497f774c0d/demo_nodes_cpp/src/parameters/parameter_events.cpp#L24) as well as [creating a logger with `rclcpp::get_logger`](https://github.com/ros2/demos/blob/e3b99c4cfdbbb50455d4a1fab17324497f774c0d/composition/src/dlopen_composition.cpp#L33) and using it for logging.

Karsten1987, 2018-10-23:
@Myzhar any updates on this?

Myzhar, 2018-10-24:
@Karsten1987 I'm sorry, but I'm not working on ROS2 now, I had to move to ROS1. 
I try to find a bit of time to complete this enhancement

Karsten1987, 2018-10-28:
fixed in https://github.com/ros2/rclcpp/commit/e30f31551e2f4731835deca51c043ae5cf099c8d

closed 2018-10-28 by Karsten1987