ros2/rclcpp #554: run 64 nodes , CPU utilization becomes 100%.

MoscowMule3, 2018-09-11:
- Operating System:
  - Ubuntu 16.04
- Installation type:
  - binaries
- Client library (if applicable):
  -  rclcpp

#### Steps to reproduce issue
On the ROS2(Ardent) environment , when I start 32 Publisher nodes and 32 Subscriber nodes, CPU utilization becomes 100%.

As  Publisher, I use the example code from 
https://github.com/ros2/examples/blob/master/rclcpp/minimal_publisher/member_function.cpp

As  Subscriber,  I use the example code from
https://github.com/ros2/examples/blob/master/rclcpp/minimal_subscriber/member_function.cpp

#### Additional information
With Python code, same issue can be reproduce.

dirk-thomas, 2018-09-11:
Have you tried the same with the latest release Bouncy? Can you share your ready-to-user example to reproduce the problem?

dirk-thomas, 2018-09-11:
I just noticed that you also posted the question here: https://answers.ros.org/question/303069/run-64-nodes-cpu-utilization-becomes-100/

Please read the ROS support guidelines which explicitly discourages cross posting the same question in different places (http://wiki.ros.org/Support#Etiquette).

Therefore I will close this ticket.

closed 2018-09-11 by dirk-thomas