ros2/rclcpp #564: LifecycleNode publishes a null timestamp in `transition_event` topics

Myzhar, 2018-10-05:
**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - Source
- Version or commit hash:
  - eb439ddc73c7f57ef59bb4919b02c4dd00d6ffec
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
1. start a lifecycle talker node: 
   `$ ros2 run lifecycle lifecycle_talker`
2. subscribe to the `transition_event` topic:
   `$ ros2 topic echo /lc_talker/transition_event`
3. require some transitions:
   `$ ros2 lifecycle set /lc_talker configure`
   `$ ros2 lifecycle set /lc_talker activate`

I noticed the issue with my own lifecycle node and I tested it with the provided example to be sure that the issue is generic.

#### Expected behavior
current timestamp for the messages
#### Actual behavior
The timestamp is null as you can see in the following screenshoot

[image]

Myzhar, 2018-10-08:
I noticed that also the `transition` field is empty.
Only the `ßtart-state` and the `goal-state` fields are filled

clalancette, 2018-10-08:
Yeah, this is because the low-level publisher only fills in `start-state` and `goal-state` here: https://github.com/ros2/rcl/blob/master/rcl_lifecycle/src/com_interface.c#L229 .  This could probably be improved, so I'm going to mark this as an enhancement.

closed 2026-02-04 by fujitatomoya