ros2/rclcpp #570: provide logging macro signature that accepts std::string

wjwwood, 2018-10-16:
## Feature request

#### Feature description

I'd like to have std::string (and possibly other common string containter) friendly logging macros.

For example, right now you must make logging calls like this:

```c++
std::string msg = "some error message from elsewhere";
RCLCPP_INFO(logger, ("some suffix: " + msg).c_str());
```

But I'd like to write:

```c++
std::string msg = "some error message from elsewhere";
RCLCPP_INFO(logger, "some suffix: " + msg);
// or just...
RCLCPP_INFO(logger, msg);
```

#### Implementation considerations

I think a helper function could make the macros simple, something like:

```c++
#include <string>

using namespace std::string_literals;

const char *
get_c_string(const char * string_in)
{
  return string_in;
}

const char *
get_c_string(const std::string & string_in)
{
  return string_in.c_str();
}

#define LOG_INFO(msg) printf("%s\n", get_c_string(msg));

int main()
{
  LOG_INFO("plain c string");
  std::string a_str("a local std::string");
  LOG_INFO(a_str);
  LOG_INFO(std::string("a temporary std::string"));
  LOG_INFO("a std::string literal"s);
}
```

Foxar, 2018-10-17:
Hello

I decided to try contributing to open source projects and since this is tagged 'good first issue' and I'm into CPP I've thought I might try doing this issue as my first.

I'm just wondering two things:
Where should I implement the macros at ? rclcpp/macros.hpp ?
Where should I implement any possible 'helper functions' like suggested here? rclcpp/utilities.hpp?

I'm sorry if I omitted something and this was specified somewhere but I can't find it anywhere.

fmrico, 2018-10-21:
Hi all,

I think I have a possible solution, in 9a99a84 

Best
Francisco

closed 2018-10-25 by wjwwood