ros2/rclcpp #571: Allocate a message with type only known at compile time

Martin-Idel-SI, 2018-10-17:
## Feature request (or request for a pointer on how to do it)

Allow allocation of message with unknown type at compile time.

#### Feature description

This issue crops up in development of rosbag2: Imagine a user created a bagfile on a machine with a certain rmw "A" and passes it to another user who wants to play it back with a second rmw "B". To allow fast writing, we need to store the messages in their serialized form, hence we need to potentially convert between the serialization format of "A" and "B" (of course, to convert both rmw's need to be present).

To allow conversion between arbitrary formats, the default interface will be to take a ROS 2 message in its serialized version and convert it to a standard ROS 2 message. In essence, this requires a call to `rmw_deserialize`.

Here is the problem:

`rmw_deserialize` takes a zero-allocated message of the correct type.

https://github.com/ros2/rmw/blob/3b9ea0f66feb161d5872cfde0317dd8280a8b91a/rmw/include/rmw/rmw.h#L250

This is no problem if we know the type at compile time. Let's say it's a `std_msgs::msg::String` so we just do
```
std_msgs::msg::String string;
rmw_deserialize(serialized_message, &load_typesupport("std_msgs/String"), &string)
```
where `load_typesupport` is some helper function loading the correct typesupport via Poco.

However, if we don't know that it's a `std_msgs::msg::String` at compile time, we can't write the first line above. In the rosbag2 scenario above, we clearly cannot know the message types at compile time, we only know them (and their typesupports) when we see the bagfile. 
Hence our feature request: It should be possible to zero-allocate a message whose type is only known at runtime.

Note that we want to support editing bagfiles with different languages in principle (e.g. python and C++), so we'd need this information for every typesupport.

#### Implementation considerations

From what I understand, this would require one of two things:
a) An information about the size of the zero-allocated message
b) A function to get a zero-allocated message (i.e. some form of constructor)

Due to name-mangling in C++, it will be very hard to get the constructor symbol out of the corresponding libraries (e.g. `libstd_msgs__rosidl_typesupport_cpp.so` for  `std_msgs/String`), so b) seems difficult.

On the other hand, this should also be a general issue for C and indeed it seems that `rclc` adds information of type a) to a new typesupport. 

https://github.com/ros2/rclc/blob/master/rclc/include/rclc/type_support.h

If I am not mistaken, this implies the size information is currently not present in the `rosidl_typesupport_t` struct. Could this be added?

Fyi @Karsten1987

Martin-Idel-SI, 2018-10-31:
After some discussion, there is a way to do this using the `rosidl_typesupport_introspection_cpp`:

The idea is to use `rosidl_introspection_typesupport_cpp` because this gives enough information about the size and type of the message to actually create it (tested on Windows, Mac and Linux). 

Here is the basic procedure:
- Zero-allocate the amount of memory given by the `size_of_` member of the introspection typesupport
- Recursively go through all members
- Overwrite the zero-allocated memory of all strings with real empty strings (this is necessary on Linux, otherwise I get segfaults when assigning an empty string at some later point)
- Overwrite the zero-allocated memory of all `std::vector<bool>` with an empty `std::vector<bool>` (this is necessary on Windows, since otherwise assigning to the vector fails)

In principle, the best way would be to overwrite every member and allocate the correct type everywhere, but I don't know how to do this for `std::vector<T>` where T is an arbitrary non-string and non-primitive type. In practice, overwriting only bool vectors and strings seems to be enough...

Later, if you want to destroy it, there is enough information in the `rosidl_introspection_typesupport_cpp` to do that: Basically you have to go through the message recursively and destroy all content and then deallocate the whole message.

If you want to use custom allocators and not the free store, I haven't yet found a good way to do this because as said above, I can't allocate a real object of type `std::vector<T>` with an arbitrary type not known at runtime.

dev-william, 2022-01-22:
I came across this while looking to do something similar. For future reference, the last couple versions of ROS2 support serialization for types known only at runtime. rosidl_typesupport_introspection_cpp::MessageMembers has init_function and fini_function to construct and destroy messages, as well as type-agnostic functions for interacting with arrays/vectors.