ros2/rclcpp #572: Adding deadline, liveliness and other QoS to ROS 2

dejanpan, 2018-10-17:
This issue is to start discussing adding of additional QoS to ROS 2. In particular the first two QoS that deem to be really necessary are:
1. **Deadline** which is needed for when data is needed at constant periodicity (e.g. vehicle data for controller at exactly 1kHz).
1. **Liveliness** which specifies and configures the mechanism that allows DDSDataReader entities to detect when DDSDataWriter entities become disconnected or "dead. This QoS setting is needed for a heartbeat feature in managed nodes.

## General approach

As ROS 2 is C++11/14 it seems natural to implement advanced features alongside the DDS C++11 API described [here](https://community.rti.com/static/documentation/connext-dds/5.2.0/doc/api/connext_dds/api_cpp2/classdds_1_1sub_1_1status_1_1DataState.html). While this API is AFAIK not an official DDS standard, or at least I could not find it on the OMG page) Connext Pro and OpenSplice use this exact API.

As most functionality is implemented around the wait-set concept it makes sense to do the same in ROS 2 as the existing callback-based API which is very similar to ROS 1 is not modified.

## DDS C++11/14 concept summary:

Everything in DDS is Event-based. Events in DDS are called **Conditions**. 
There are several kinds of Conditions:

#### [Read Conditions](https://community.rti.com/static/documentation/connext-dds/5.2.0/doc/api/connext_dds/api_cpp2/classdds_1_1sub_1_1status_1_1InstanceState.html)

Everything that has to do with the *state* of a sample or instance (keyed group).
A read condition is triggered when a sample with the attached (Data State)[https://community.rti.com/static/documentation/connext-dds/5.2.0/doc/api/connext_dds/api_cpp2/classdds_1_1sub_1_1status_1_1DataState.html] is received or an existing sample changes to this data state.

#### [Status Condition](https://community.rti.com/static/documentation/connext-dds/5.2.0/doc/api/connext_dds/api_cpp2/classdds_1_1core_1_1cond_1_1StatusCondition.html)

Everything that is related to events not to a state of a sample. 
**Examples:**
New data reader/writer detected.
Incompatible QOS detected.
Deadline missed.
DataWriter died.

List of possible status: https://community.rti.com/static/documentation/connext-dds/5.2.0/doc/api/connext_dds/api_cpp2/group__DDSStatusTypesModule.html

## Condition management:

The user needs to be able to check if a condition did happen. There are two ways to do this:
* Waitsets: The recommended way by RTI.
* Callbacks: Only recommended for special ultra-low latency applications (can be dangerous) (by RTI)[https://community.rti.com/best-practices/use-waitsets-except-when-you-need-extreme-latency]. Also, the callback based approach is much harder to use in my experience.

So `rclcpp` should implement the waitset based approach:
https://community.rti.com/static/documentation/connext-dds/5.2.0/doc/api/connext_dds/api_cpp2/classdds_1_1core_1_1cond_1_1WaitSet.html

## QOS Configuration

QOS settings must also be configured, which is the easier thing to do.

**Deadline:**
https://community.rti.com/static/documentation/connext-dds/5.2.0/doc/api/connext_dds/api_cpp2/classdds_1_1core_1_1policy_1_1Deadline.html

**Ownership:**
https://community.rti.com/static/documentation/connext-dds/5.2.0/doc/api/connext_dds/api_cpp2/group__DDSOwnershipQosModule.html
https://community.rti.com/static/documentation/connext-dds/5.2.0/doc/api/connext_dds/api_cpp2/group__DDSOwnershipStrengthQosModule.html

**Liveliness**
https://community.rti.com/static/documentation/connext-dds/5.2.0/doc/api/connext_dds/api_cpp2/group__DDSLivelinessQosModule.html

## Utility functions

Some utility functions like asserting liveliness and translating instance handles to keys and via versa must also be provided on a DataWriter and DataReader level.

dejanpan, 2018-11-15:
Update:
## Description

Because of the following considerations, it is desired to have direct access to Connext Micro DDS entries from ROS 2.
* The timeline for advanced QOS settings (#2094) is not clear and it is not clear that all the required features will be finished for Apex.OS release.
* The additional overhead ROS 2 introduces by copying could be disqualifying for some customers and applications. A very rough estimation of this overhead is given in https://gitlab.apex.ai/ApexAI/grand_central/issues/2094#note_119982.
* It is unclear if a unifying type abstraction is possible by working ROS 2 and DDS types at the same time. A clean solution requires additional ROS 2 functionality: https://gitlab.apex.ai/ApexAI/grand_central/issues/2131#note_118771

## API Requirement.

The informations in this section are derived from the Connext Micro Documentation: https://community.rti.com/static/documentation/connext-micro/2.4.6/doc/html/group__CPPApiModule__SupportedAPI.html

There is a hirachical structure of DDS entities:

* Participant
    * Subscriber
    * Publisher
        * DataReader
        * DataWriter

All DDS entities allow to get the hirachical higher level entity. For example.

```
   DDSDataReader dr;
   DDSSubscriber sub = dr.get_subscriber();
   DDSParticipant dp = sub.get_participant();
```

Therefore `rclcpp` only needs to provide access to DDS `DataReaders` and `DataWriters`.

An API could look like:

```
rclcpp::Node mynode(...);
auto sub = mynode.get_subscription<some_ros_type>(..., disabled=true); 
auto ddsdr = sub.dds(); // DataReader, rest of the hirachy can be reached as described above.
```

A mirroring API for the DataWriter should also be implemented.

**As shown in the above proposal there must be a way to start DDS entities disabled as it is not possible to change some settings anymore after they got enabled.**

## Type Support

There must exist a matching between ROS types and DDS types in form of a trait or similar:
```
template<class ROS_TYPE>
class ros_dds_trait {};

template<>
class ros_dds_trait<some_ros_type> {
public:
  using dds_type_t = some_ros_type
  static auto * ConnextDDSMicroTypePlugin()
  {
    return dds_type_t_TypePlugin_get();
  }
};
```

How these traits are generated is to define. The most obvious place would be the ROS 2 type generation.

The usage would be then as follows:

```
rclcpp::Node mynode(...);
auto sub = mynode.get_subscription<some_ros_type>(..., disabled=true); 
auto ddsdr = sub.dds(); // DataReader, rest of the hirachy can be reached as described above.
auto typed_dds_dr = class ros_dds_trait<some_ros_type>::DataReader::narrow(ddsdr);
```

Best is to additionally support directly a function on the ROS 2 subscription which handles this for the user:
```
auto typed_dds_dr = sub.dds_typed();
```

I think the untyped type also needs to be provided as only it has the full API.

dejanpan, 2019-03-04:
@deeplearningrobotics why did we bring Ownership QoS in this issue as well?

deeplearningrobotics, 2019-03-04:
@dejanpan: Because it is required to build a redundant system.

gbiggs, 2019-03-04:
Yes, you need ownership to manage which publishers are in redundant fail-over groups. It's relevant both to safety and security.

wjwwood, 2019-09-24:
Can you guys rework this issue (close it and open a new one maybe?), since deadline, liveliness, and lifespan were added, but ownership still has not been added.

wjwwood, 2020-03-23:
I'm going to close this as stale, please feel free to open a new issue for tracking the Ownership QoS feature.

closed 2020-03-23 by wjwwood