ros2/rclcpp #580: ros time source should not use parameter client

wjwwood, 2018-11-07:
Right now the TimeSource class implementation in rclcpp uses a ParameterClient to get notifications of when a parameter changes, which:

- does not work if parameter service is disabled in the node
- is not required, the node knows when the parameter changes (with or without ROS services)

wjwwood, 2018-11-07:
https://github.com/ros2/rclcpp/blob/33a755c535d654c97401eb6d199580368c19eb40/rclcpp/src/rclcpp/time_source.cpp#L69-L74

CursedRock17, 2023-09-29:
Since pull request #740 was merged, `time_source` no longer uses `parameter_client_`, so this issue should have been solved correct? Otherwise I'd happy to make some changes needed.

clalancette, 2023-10-02:
Yeah, I'm going to close this one out for now.  We may have to make further changes to the TimeSource (see the conversation in https://github.com/ros2/geometry2/pull/630), but that is not what this PR is talking about.

closed 2023-10-02 by clalancette