ros2/rclcpp #583: No matching function build error when adding arguments to subscriber callback

alsora, 2018-11-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - from source
- Version or commit hash:
  - 0.5.1
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
```
#include "rclcpp/rclcpp.hpp"
#include "std_msgs/msg/header.hpp"
#include <string>
#include <iostream>

class MyNode : public rclcpp::Node{
public:

    MyNode() : Node("test_node")
    {
    }

    void callback(const std_msgs::msg::Header::SharedPtr msg, std::string topic_name)
    {
        (void)msg;
        std::cout<<topic_name<<std::endl;
    }
    rclcpp::Subscription<std_msgs::msg::Header>::SharedPtr subscriber;
};

int main(int argc, char const *argv[])
{
    rclcpp::init(argc, argv);

    std::string topic_name = "test_topic";

    auto node = std::make_shared<MyNode>();
    node->subscriber = node->create_subscription<std_msgs::msg::Header>(
        topic_name,
        std::bind(&MyNode::callback, node, std::placeholders::_1, topic_name));

    rclcpp::spin(node);

    rclcpp::shutdown();

    return 0;
}
```
#### Actual behavior

The provided code does not compile due to a `no matching function` error while creating the subscription.

#### Additional information

Original ROS answers question for this bug 

https://answers.ros.org/question/308386/ros2-add-arguments-to-callback/

mjcarroll, 2018-11-27:
One option would be to use a lambda.

```    
node->subscriber = node->create_subscription<std_msgs::msg::Header>(
        topic_name,
        [node, topic_name](const std_msgs::msg::Header::SharedPtr msg) {
          node->callback(msg, topic_name);
        });
```

mjcarroll, 2018-11-27:
Additionally, it seems like there may be some type deduction issues here, if you explicitly type, it also works:

```
  std::function<void(const std_msgs::msg::Header::SharedPtr)> callback = std::bind(&MyNode::callback,
      node.get(), std::placeholders::_1, topic_name);

  node->subscriber = node->create_subscription<std_msgs::msg::Header>(topic_name, callback);
```

wjwwood, 2018-11-27:
@mjcarroll see the ros answers post

Karsten1987, 2019-09-10:
I believe that's a duplicate of #273 which has an answer here: https://github.com/ros2/rclcpp/issues/273#issuecomment-263826519

dirk-thomas, 2019-09-10:
To summarize: I think the failing case is not supported since we can't ensure at compile time that the signature using `std::bind` is correct. Therefore I will go ahead and close this ticket.

For reference: #766 enumerates a few cases - some which work - some which don't.

Please feel free to continue commenting on it and it can be reopened if necessary.

closed 2019-09-10 by dirk-thomas