ros2/rclcpp #591: add configurations for parameter events

wjwwood, 2018-11-27:
It should be possible to:

- disable the parameter event publisher
- control the QoS settings for the parameter event publisher
  - See: https://github.com/ros2/rclcpp/blob/27b0428f7a46d95c1f39494b2c2d98df04bd5576/rclcpp/src/rclcpp/node_interfaces/node_parameters.cpp#L56

rarvolt, 2018-12-29:
First time contributing to this project. I'd like to take this issue and already written some code. Should I create Pull Request to this issue to discuss changes or publish them somewhere else?

dirk-thomas, 2018-12-29:
> Should I create Pull Request to this issue to discuss changes ...

Absolutely, please go ahead and create a pull request.

wjwwood, 2019-03-29:
I believe this is done as of #614.

closed 2019-03-29 by wjwwood