ros2/rclcpp #595: 'use_sim_time' only works when set dynamically, not through launch parameters

mkhansen-intel, 2018-11-29:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source using master ros2.repos file
- Version or commit hash:
  - I'll get this if you need it
- DDS implementation:
  - Fast-RTPS (default)
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Launch file:
```
from launch import LaunchDescription

import launch.actions
import launch_ros.actions

def generate_launch_description():
    return LaunchDescription([

        launch.actions.ExecuteProcess(
            cmd=['gzserver', '--verbose', '-s', 'libgazebo_ros_init.so', 'empty.world'],
            output='screen'),

        launch_ros.actions.Node(
            package='testpackage',
            node_executable='testpackage',
            node_name='minimal_subscriber',
            output='screen',
            parameters=[{'use_sim_time': True}])
    ])
```
Source file:
```
#include <memory>
#include <exception>
#include "rclcpp/rclcpp.hpp"

int main(int argc, char ** argv)
{
  rclcpp::init(argc, argv);
  try {
    auto node=rclcpp::Node::make_shared("minimal_subscriber");
    rclcpp::Rate loop_rate(1);
    while(rclcpp::ok()) {
      RCLCPP_INFO(node->get_logger(), "Node Time is %ld", node->now().nanoseconds());
      RCLCPP_INFO(node->get_logger(), "Clock Time is %ld", rclcpp::Clock(RCL_ROS_TIME).now().nanoseconds());
      rclcpp::spin_some(node);
      loop_rate.sleep();
    }
  } catch (...) {
    std::cerr << "caught exception" << std::endl;
  }
  rclcpp::shutdown();

  return 0;
}
```
#### Expected behavior
When I set 'use_sim_time' through a launch file, it should start using the simulation clock from gazebo as soon as it is available. This is seen in the output below as 'Node Time'

#### Actual behavior
The node won't actually switch to sim time unless use_sim_time is set dynamically after the node has launched. 

#### Additional information
```
mkhansen@mkhansen-desk:~/ros2_dev/navigation2_ws$ ros2 launch testpackage testpackage_launch.py
[INFO] [launch]: process[gzserver-1]: started with pid [26616]
[INFO] [launch]: process[testpackage-2]: started with pid [26617]
[INFO] [minimal_subscriber]: Node Time is 1543457809023725948
[INFO] [minimal_subscriber]: Clock Time is 1543457809023744824
Gazebo multi-robot simulator, version 9.0.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.7.201.25
[Err] [RenderEngine.cc:725] Can't open display:
[Wrn] [RenderEngine.cc:93] Unable to create X window. Rendering will be disabled
[Wrn] [RenderEngine.cc:293] Cannot initialize render engine since render path type is NONE. Ignore this warning ifrendering has been turned off on purpose.
[Err] [Server.cc:379] Could not open file[empty.world]
[Wrn] [Server.cc:338] Falling back on worlds/empty.world
[INFO] [minimal_subscriber]: Node Time is 1543457810023793285
[INFO] [minimal_subscriber]: Clock Time is 1543457810023806356
[INFO] [minimal_subscriber]: Node Time is 1543457811023812189
[INFO] [minimal_subscriber]: Clock Time is 1543457811023822896
[INFO] [minimal_subscriber]: Node Time is 1543457812023810425
[INFO] [minimal_subscriber]: Clock Time is 1543457812023822080
[INFO] [minimal_subscriber]: Node Time is 1543457813023811638
[INFO] [minimal_subscriber]: Clock Time is 1543457813023824679
[INFO] [minimal_subscriber]: Node Time is 1543457814023787142
[INFO] [minimal_subscriber]: Clock Time is 1543457814023800707
[INFO] [minimal_subscriber]: Node Time is 1543457815023810150
[INFO] [minimal_subscriber]: Clock Time is 1543457815023821468
[INFO] [minimal_subscriber]: Node Time is 1543457816023783249
[INFO] [minimal_subscriber]: Clock Time is 1543457816023795940
[INFO] [minimal_subscriber]: Node Time is 1543457817023781949
[INFO] [minimal_subscriber]: Clock Time is 1543457817023793476
[INFO] [minimal_subscriber]: Node Time is 1543457818023808947
[INFO] [minimal_subscriber]: Clock Time is 1543457818023820761
[INFO] [minimal_subscriber]: Node Time is 1543457819023784617
[INFO] [minimal_subscriber]: Clock Time is 1543457819023798860
[INFO] [minimal_subscriber]: Node Time is 1543457820023815570
[INFO] [minimal_subscriber]: Clock Time is 1543457820023829309
[INFO] [minimal_subscriber]: Node Time is 1543457821023812025
[INFO] [minimal_subscriber]: Clock Time is 1543457821023824201
[INFO] [minimal_subscriber]: Node Time is 1543457822023811756
[INFO] [minimal_subscriber]: Clock Time is 1543457822023824277
[INFO] [minimal_subscriber]: Node Time is 1543457823023811181
[INFO] [minimal_subscriber]: Clock Time is 1543457823023824359
[INFO] [minimal_subscriber]: Node Time is 1543457824023806113
[INFO] [minimal_subscriber]: Clock Time is 1543457824023818997
[INFO] [minimal_subscriber]: Node Time is 1543457825023780745
[INFO] [minimal_subscriber]: Clock Time is 1543457825023792874
[INFO] [minimal_subscriber]: Node Time is 1543457826023785984
[INFO] [minimal_subscriber]: Clock Time is 1543457826023799544
[INFO] [minimal_subscriber]: Node Time is 1543457827023784097
[INFO] [minimal_subscriber]: Clock Time is 1543457827023795893
[INFO] [minimal_subscriber]: Node Time is 1543457828023810237
[INFO] [minimal_subscriber]: Clock Time is 1543457828023824484
[INFO] [minimal_subscriber]: Node Time is 1543457829023804999
[INFO] [minimal_subscriber]: Clock Time is 1543457829023820734
[INFO] [minimal_subscriber]: Node Time is 1543457830023813067
[INFO] [minimal_subscriber]: Clock Time is 1543457830023825471
[INFO] [minimal_subscriber]: Node Time is 1543457831023804823
[INFO] [minimal_subscriber]: Clock Time is 1543457831023817486
[INFO] [minimal_subscriber]: Node Time is 1543457832023810373
[INFO] [minimal_subscriber]: Clock Time is 1543457832023820863
[INFO] [minimal_subscriber]: Node Time is 1543457833023812489
[INFO] [minimal_subscriber]: Clock Time is 1543457833023824741
[INFO] [minimal_subscriber]: Node Time is 0
[INFO] [minimal_subscriber]: Clock Time is 1543457834023820720
[INFO] [minimal_subscriber]: Node Time is 24576000000
[INFO] [minimal_subscriber]: Clock Time is 1543457835023791415
[INFO] [minimal_subscriber]: Node Time is 25581000000
[INFO] [minimal_subscriber]: Clock Time is 1543457836023819240
[INFO] [minimal_subscriber]: Node Time is 26584000000
[INFO] [minimal_subscriber]: Clock Time is 1543457837023795041
[INFO] [minimal_subscriber]: Node Time is 27588000000
[INFO] [minimal_subscriber]: Clock Time is 1543457838023820281
[INFO] [minimal_subscriber]: Node Time is 28589000000
[INFO] [minimal_subscriber]: Clock Time is 1543457839023821805
[INFO] [minimal_subscriber]: Node Time is 29492000000
[INFO] [minimal_subscriber]: Clock Time is 1543457840023791436
[INFO] [minimal_subscriber]: Node Time is 30497000000
[INFO] [minimal_subscriber]: Clock Time is 1543457841023795913
[INFO] [minimal_subscriber]: Node Time is 31500000000
[INFO] [minimal_subscriber]: Clock Time is 1543457842023852300
[INFO] [minimal_subscriber]: Node Time is 32503000000
[INFO] [minimal_subscriber]: Clock Time is 1543457843023825404
[INFO] [minimal_subscriber]: Node Time is 33505000000
[INFO] [minimal_subscriber]: Clock Time is 1543457844023821935
[INFO] [minimal_subscriber]: Node Time is 34509000000
[INFO] [minimal_subscriber]: Clock Time is 1543457845023784820
[INFO] [minimal_subscriber]: Node Time is 35511000000
[INFO] [minimal_subscriber]: Clock Time is 1543457846023825233
```
Here you can see the parameter was set already by the launch file, thus the param 'get' returned True. But the node time didn't change until I set it a 2nd time dynamically.
```
mkhansen@mkhansen-desk:~/ros2_dev/navigation2_ws$ ros2 param get /minimal_subscriber use_sim_time
Boolean value is: True
mkhansen@mkhansen-desk:~/ros2_dev/navigation2_ws$ ros2 param get /minimal_subscriber use_sim_time
Boolean value is: True
mkhansen@mkhansen-desk:~/ros2_dev/navigation2_ws$ ros2 param set /minimal_subscriber use_sim_time True
Set parameter successful
```
Credit to @crdelsey who helped to find / isolate this bug and wrote the test node

tfoote, 2018-12-05:
I cannot reproduce the bug as described requiring user interaction. I do find a small race condition on startup that the clock does not get initialized until after the node has been spun at least once. 

```
[INFO] [launch]: process[publisher_member_function-1]: started with pid [11375]
[INFO] [publisher_member_function]: Before first spin
[INFO] [publisher_member_function]: - Node Time is 1543995895338527930
[INFO] [publisher_member_function]: - Clock Time is 1543995895338535193
[INFO] [publisher_member_function]: After first spin
[INFO] [publisher_member_function]: Count: 1
[INFO] [publisher_member_function]: - param: true
[INFO] [publisher_member_function]: - Node Time is 0
[INFO] [publisher_member_function]: - Clock Time is 1543995895341217242
[INFO] [publisher_member_function]: Count: 2
[INFO] [publisher_member_function]: - param: true
[INFO] [publisher_member_function]: - Node Time is 10000000001000
[INFO] [publisher_member_function]: - Clock Time is 1543995896336628316
[INFO] [publisher_member_function]: Count: 3
[INFO] [publisher_member_function]: - param: true
[INFO] [publisher_member_function]: - Node Time is 10000000001000
[INFO] [publisher_member_function]: - Clock Time is 1543995897336554403
```

Running the slightly modified program https://github.com/tfoote/bug595/blob/master/bug_595.cpp

Note that the commented out get_parameter in the before spin section will have the side effect of causing the parameter value to be evaluated. https://github.com/tfoote/bug595/blob/8bca0a2c7a33dee53a247b8c12a669d5e24bd1d4/bug_595.cpp#L17

Note that I was substituting for gazebo a rostopic publisher sending `/clock` though I did not see that change behavior with it present or not.https://github.com/tfoote/bug595/blob/8bca0a2c7a33dee53a247b8c12a669d5e24bd1d4/test.launch.py#L11

Without the parameter set in the launch file I get the following expected behavior.
```
[INFO] [launch]: process[publisher_member_function-1]: started with pid [11321]
[INFO] [publisher_member_function]: Before first spin
[INFO] [publisher_member_function]: - Node Time is 1543995880360432223
[INFO] [publisher_member_function]: - Clock Time is 1543995880360440390
[INFO] [publisher_member_function]: After first spin
[INFO] [publisher_member_function]: Count: 1
[INFO] [publisher_member_function]: - param: false
[INFO] [publisher_member_function]: - Node Time is 1543995880362001382
[INFO] [publisher_member_function]: - Clock Time is 1543995880362004867
[INFO] [publisher_member_function]: Count: 2
[INFO] [publisher_member_function]: - param: false
[INFO] [publisher_member_function]: - Node Time is 1543995881358593778
[INFO] [publisher_member_function]: - Clock Time is 1543995881358598954
[INFO] [publisher_member_function]: Count: 3
[INFO] [publisher_member_function]: - param: false
[INFO] [publisher_member_function]: - Node Time is 1543995882358641954
[INFO] [publisher_member_function]: - Clock Time is 1543995882358647313
[INFO] [publisher_member_function]: Count: 4
[INFO] [publisher_member_function]: - param: false
[INFO] [publisher_member_function]: - Node Time is 1543995883358537995
[INFO] [publisher_member_function]: - Clock Time is 1543995883358543539
```

I believe that we should make sure that the parameters get propogated into the node on startup before a spin is called if they are set prior to node startup. But without being able to reproduce the full latency going until the user interaction I don't know what to debug. My only speculation is that somehow it's a thread starvation where the spinning doesn't get to the parameter callbacks because it's running other callbacks etc.

@mkhansen-intel @crdelsey do you see this behavior with this sample package?
 Or can you modify it to demonstrate the behavior described? https://github.com/tfoote/bug595

tfoote, 2018-12-06:
I think that this is fundamentally a race condition at startup which explains why we're seeing it behave differently under different circumstances (such as the nav stack). The code currently relies on creating a subscription when a node is attached: https://github.com/ros2/rclcpp/blob/33f1e1776c63a37623c0a1f006b4951e4522f071/rclcpp/src/rclcpp/time_source.cpp#L81 I verified that when I run bug595 this callback is hit. However I believe that if the TimeSource is initialized later (programmatically) or the machine is under high load/context switching(or another executor thread) it would be possible to have the parameter event happen before the callback is registered.

This would explain why it has to be manually triggered by an external state change. The event will get fired and the time_source will synchronize with the node's parameter when the new set param call gets executed. Usually the call would be a no-op but since the time_source is not initialized with the state of the node it updates the time_source. 

The "simple" solution would be to synchronously query the parameter state after declaring the event callback. So at worst you would get a duplicate callback for an event occurring between registering the callback and reading the parameter. And you wouldn't miss any state. However we don't have an executor in this context to use to spin/wait on for the synchronous behavior as is done in the synchronous parameter client. The clock interface is also initialized before the parameter interface now because the parameter interface uses the clock interface to send events.

crdelsey, 2018-12-06:
The bug595 code consistently seems to work now, but the nav2 stack seems to be consistently failing still.

I'll have to try some experiments. In some cases the observable time is set in the constructor before the first spin, in other cases, the nodes with observable times are composed nodes, or are in namespace.

@tfoote From your description, it sounds like the TimeSource does not need to call `get_parameter_or` or equivalent. The expected behavior is that some startup code somewhere will look at all the parameters on a node at startup and send an initial parameter event with those values?

tfoote, 2018-12-08:
So I've now dug into how this is effecting things and as @chapulina discovered there's actually two cycles necessary for the clock to get updated. 

The underlying flow of events is that:

1. The node clock interface is created, it subscribes to parameter events
2. The node parameter interface is created, it reads the parameters and publishes the parameter changes
3. After the node gets a chance to spint the node clock interface updates based on the parameter events and subscribes to the clock
4. After some discovery time and some spinning happening again the clock messages start to come it.

The underlying issue is that the clock is relying on the published events and cannot react until the node is spun to first get the parameter and then the clock. If the clock entity could check the parameter at startup and immediately subscribe the behavior would be much more intuitive. Also we should make it so that the node internals do not require external communications for this sort of configuration changes.

The obvious solution is to give the clock a pointer to the parameter interface, but that make a loop because the parameter interface already has a pointer to the clock. So to clean this up we can follow the same pattern we're already using to separate the clock from the time source. Such that the clock is just keeping the time, and the parameter can reference it appropriately. And a new time_source interface can query the parameter, subscribe to the clock topic and update the clock when appropriate. To do this we will need a new time_source_interface, which will just split out the clock and time_source datatypes already contained in the clock_interface. 

The second level of improvement would be that the parameter_interface be extended to provide a way to register a callback for changes to the parameters so that the time_source can get callbacks without relying on external communications.

Once that callback functionality is provided the parameter event publishing can actually be separated into another component that can use the callback notifications to publish the events externally. This would also have the benefit of breaking the dependency of the parameters on the clock. And lastly it could be leveraged to deal with startup conditions to defer the publishing of events until the clock is known to be initialized to avoid all startup parameters always being published at walltime.

Thanks to @wjwwood for sketching this through with me. Here's our notes from the whiteboard.

[image]

tfoote, 2018-12-11:
If anyone can test #608 with the navigation stack that would be appreciated. @mkhansen-intel @crdelsey

crdelsey, 2018-12-11:
Way to go! That did the trick. Everything is coming up with sim_time_enabled now. :-)

PR #608 (2018-12-12): Resolve startup race condition for sim time
Fixes #595 

A race condition at startup was identified in #595 where the clock and parameter interfaces needed each other to initialize and since they could not rely on both being initialized the loop was closed through sending and receiving ROS messages. This however meant that initialization was dependent on messages being passed through ROS topics before time would be initialized correctly. The solution to this is to break apart the clock dependency into the time source and clock separately, like we already do for the underlying data and this bre [... cut]
commit c93beb5d (2018-12-12): Resolve startup race condition for sim time (#608)

Resolves #595 

* Separate the Node Time Source from the Node Clock
* Implement initial value checking of use_sim_time parameter parameter
* Be sure to update all newly attached clocks
* Homogenizing the behavior to use the last received value otherwise zero time when enabling sim time.
* Add virtual destructors to interface classes
closed 2018-12-12 by tfoote