ros2/rclcpp #599: SyncParametersClient creation for a class inheriting from rclcpp::Node

alsora, 2018-12-04:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - from binary
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
```
#include "rclcpp/rclcpp.hpp"

class MyNode : public rclcpp::Node {

public:

    MyNode()
    {
        _async_param_client = std::make_shared<rclcpp::AsyncParametersClient>(this, "other_node");
        _sync_param_client = std::make_shared<rclcpp::SyncParametersClient>(this, "other_node");
    }

private:

    rclcpp::SyncParametersClient::SharedPtr _sync_param_client;
    rclcpp::AsyncParametersClient::SharedPtr _async_param_client;
};

```

#### Expected behavior

I create a new class which inherits from `rclcpp::Node` and I want to create synchronous and asynchronous parameter clients.

#### Actual behavior

The syncrhonous parameter client creation fails due to 

> /usr/include/c++/7/ext/new_allocator.h:136:4: error: no matching function for call to ‘rclcpp::SyncParametersClient::SyncParametersClient(MyNode*, const char [11])’
>   { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }

On the other hand the asynchronous parameter client works.

It still do not work if I pass as argument `this->get_node_base_interface()`

clalancette, 2018-12-06:
Right, it looks like that on bouncy, we just don't have an overloaded `SyncParametersClient` constructor that takes in an `rclcpp::Node *`, while we do have one for `AsyncParametersClient`.  I'll mark this as an enhancement, though it is a fairly low priority for us at the moment.  If you have time to add that constructor, that would move it along faster.  Thanks.

alsora, 2018-12-10:
Ok perfect, I will make a PR very soon.

wjwwood, 2018-12-12:
You could use `this->shared_from_this()`, but that won't work in the constructor.

So a raw pointer version would be good.

closed 2018-12-20 by alsora