ros2/rclcpp #609: Provide for local callbacks on parameter events in parameter node API

tfoote, 2018-12-11:
This is a follow up to discussions in #595 and #608 and somewhat related to #296 but for the post change callbacks not validation stage which would be a 3rd callback type proposed.

It would be valuable to not rely on parameter events being published but use the internal node interface API to get the events in a callback for different interfaces. Along with that change if there was a more fully featured API for registering callbacks for parameter changes locally, the parameter event publishing could also be split out into a separate component that could obviate the need for a parameter to turn that functionality on and off in the constructor of the parameter, and avoid the dependencies on the clock and topics interface for the core parameter API.

0-admin, 2019-04-15:
Wrere ti puch? kjk?(

clalancette, 2023-09-27:
This was actually completed for Iron, so closing this out.

closed 2023-09-27 by clalancette

jrutgeer, 2023-11-14:
@clalancette 
It seems this issue is not fully complete, see [this TODO item in the code](https://github.com/ros2/rclcpp/blob/411dbe8212dd753ee60871f8605d0a8beaae5e37/rclcpp/src/rclcpp/time_source.cpp#L280).

I came across this due to someone asking about disabling `/parameter_events` subscribers [on Robotics Stack Exchange](https://robotics.stackexchange.com/questions/105382/how-to-stop-the-subscription-to-parameter-events-that-is-automatically-create).

So I think this should be reopened.

clalancette, 2023-11-14:
> It seems this issue is not fully complete, see [this TODO item in the code](https://github.com/ros2/rclcpp/blob/411dbe8212dd753ee60871f8605d0a8beaae5e37/rclcpp/src/rclcpp/time_source.cpp#L280).

This issue is indeed complete; we now provide a local callback that can be used.  If you'd like to track changing the code in `time_source.cpp` to use that, please feel free to open a new issue.

fujitatomoya, 2023-11-14:
@jrutgeer to be honest, i do not remember... probably i was working on the experimental patch? bad memory, sorry 😓 

created tracking issue for this, https://github.com/ros2/rclcpp/issues/2370