ros2/rclcpp #620: ROS2 parameter with component is not working

poonam1120, 2019-01-21:
## Bug report
Use of component and ROS2 parameter is not working together

**Required Info:**
Trying to modify the demo example of composition in order to get set/get parameters from different nodes. Set the parameters on Talker component and get/list the parameter on the Listener component.
But it s not hap

- Operating System:
 Ubuntu 18.04

- Installation type:
  Using ROS2 crystal

- Version or commit hash:
NA

- DDS implementation:
 NA

- Client library (if applicable):
   rclcpp

#### Steps to reproduce issue
With demo example of composition,
Try to set the parameter client for Talker component as well as Listener component.
Want to get the parameters set by Talker on the Listener

#### Expected behavior
Parameter client should be created in Talker component and should be able to set the parameters.

#### Actual behavior
Not able to set the parameter client, getting error with “rclcpp::SyncParametersClient::SyncParametersClient”. Because this is reference to talker node which is of type composition.

#### Additional information
Not able to create the parameter client for a node which is of type "composition"

dirk-thomas, 2019-02-06:
Please include more information. The following sequence works just fine for me with Crystal:

```
ros2 run composition api_composition
# another shell:
ros2 run composition api_composition_cli composition composition::Talker
ros2 run composition api_composition_cli composition composition::Listener
ros2 param list
ros2 param set /talker foo bar
ros2 param set /listener ping pong
ros2 param list
```

poonam1120, 2019-02-07:
Thanks @dirk-thomas for your response.
Yes, it worked from command line.
Actually I want to do it from source files as I want to achieve dynamic reconfiguration(as in ros2 parameter feature) on components. Set parameters in talker_component.cpp and get/list parameters in listener_component.cpp
When I try to do, first I try to create parameter client for talker, where it throws errors that "talker node  is of type composition" and same for listener.
Hope this makes my doubt clear.
Thanks.

dirk-thomas, 2019-02-07:
> it throws errors that "talker node is of type composition"

Can you please share a reproducible example a well as the exact error message you are seeing.

poonam1120, 2019-02-18:
Please find the below code file where I am trying to create parameter client for node Talker.
[talker_component.txt](https://github.com/ros2/rclcpp/files/2874101/talker_component.txt)

Error:-

> /home/user/composition/src/talker_component.cpp:45:80:   required from here
> /usr/include/c++/7/ext/new_allocator.h:136:4: error: no matching function for call to ‘rclcpp::SyncParametersClient::SyncParametersClient(composition::Talker*)’
>   { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
>     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Thanks.

dirk-thomas, 2019-02-19:
@poonam1120 Please note that cross posting your questions is against the [etiquette of the project](http://wiki.ros.org/Support#Etiquette): https://discourse.ros.org/t/composition-and-parameters-best-practice-suggestions/1001/13

Please try the suggestion from this comment: https://github.com/ros2/rclcpp/issues/599#issuecomment-446756330

dirk-thomas, 2019-02-26:
Since there was no further reply I assume the problem was addressed by the previous comments.

closed 2019-02-26 by dirk-thomas