ros2/rclcpp #623: Expose matched subscriber count for publishers

paulbovbel, 2019-01-30:
## Feature request

#### Feature description
Please expose a mechanism for getting matched pub/sub counts (rcl_publisher_get_subscription_count).

#### Implementation considerations
I was catching up on https://answers.ros.org/question/286069/implementing-subscriberstatuscallback-within-ros2/ and noticed that `Expose matched publisher / subscriber count (rather than only based on the topic name)` is no longer in the roadmap! It looks like rcl_publisher_get_subscription_count was imlemented in rcl/rmw but never plumbed out to the client libraries. Is there any reason for that?

Also would there be any way to get that data in a callback/async rather than by polling?

mjcarroll, 2019-01-30:
Yes, it didn't quite get all the way up into the `rclcpp` library, but it hasn't been completely dropped.  It has been partially rolled in to: https://github.com/ros2/ros2/issues/649

As far as callback/async, it is possible, just currently not implemented.  It would work similar to how incoming messages work in `rcl`, with a corresponding mapping in `rclcpp`.

mjcarroll, 2019-01-30:
Agreed.

The approach would be to use another guard condition (or perhaps piggy-back off an existing one) to trigger when publisher and subscriber counts change.  This could then be used at the `rclcpp`-layer to provide the behavior you describe.

ivanpauno, 2019-03-13:
This was solve in https://github.com/ros2/rclcpp/pull/628, I forgot referencing this

paulbovbel, 2019-03-13:
Thanks @ivanpauno!

Is adding an async mechanism for notifying about new subscribers still on the roadmap for rclcpp, to enable lazy pub/sub pipelines?

dirk-thomas, 2019-03-13:
> This was solve in #628

Closing...

> Is adding an async mechanism for notifying the application regarding new subscribers still on the roadmap for rclcpp

It is mentioned as a desired feature on the [roadmap](https://index.ros.org/doc/ros2/Roadmap/#new-features) but it is not planned to work on it by [Dashing](https://github.com/ros2/ros2/issues/607).

closed 2019-03-13 by dirk-thomas