ros2/rclcpp #624: Problems with client.hpp "wait_for_service" function using Clang

kuzai, 2019-01-30:
## Bug report
rclcpp won't compile using a 64-bit architecture on Clang++
**Required Info:**

- Operating System:
  - OSX Mojave
- Installation type:
  - Source
- Version or commit hash:
  - 0.5.1, but looks to be present in newer versions
- DDS implementation:
  Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Attempt to compile rclcpp using 64-bit clang

#### Expected behavior

Compiles 

#### Actual behavior

Error thrown, type error between long and long long when parameter_client.cpp calls client->wait_for_service()

#### Additional information

https://stackoverflow.com/questions/36302993/clangs-libc-using-long-long-in-the-definition-of-stdchronoduration

Looks to be a clang specific issue with regards to how it implements std::chrono

clalancette, 2019-01-30:
We're currently only supporting Sierra with ROS 2: https://index.ros.org/doc/ros2/Installation/OSX-Install-Binary/ .

The main reason is that we ran into a lot of performance issues on High Sierra, and there is a bunch of work to do to make ROS 2 work on Mojave.  That being said, t's fairly low priority for us to move to Mojave, so if you are interested in getting ROS 2 to work on Mojave, pull requests would be happily reviewed.

wjwwood, 2020-03-23:
Closing since we've upgraded to new macOS since then.

closed 2020-03-23 by wjwwood