ros2/rclcpp #625: Ros2 Service Send Request Does Not Send

borgmanJeremy, 2019-01-30:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
  - Docker instance based on 18.04  
- Installation type:
  - source
- Version or commit hash:
ros2.repos:
```
repositories:
  ament/ament_cmake:
    type: git
    url: https://github.com/ament/ament_cmake.git
    version: 0.6.0
  ament/ament_index:
    type: git
    url: https://github.com/ament/ament_index.git
    version: 0.5.1
  ament/ament_lint:
    type: git
    url: https://github.com/ament/ament_lint.git
    version: 0.6.2
  ament/ament_package:
    type: git
    url: https://github.com/ament/ament_package.git
    version: 0.6.0
  ament/googletest:
    type: git
    url: https://github.com/ament/googletest.git
    version: c8b6f2be853002b9a0abfb30ea537402383bf1f4
  ament/osrf_pycommon:
    type: git
    url: https://github.com/osrf/osrf_pycommon.git
    version: 0.1.6
  ament/uncrustify_vendor:
    type: git
    url: https://github.com/ament/uncrustify_vendor.git
    version: 1.0.0
  eProsima/Fast-CDR:
    type: git
    url: https://github.com/eProsima/Fast-CDR.git
    version: v1.0.8
  eProsima/Fast-RTPS:
    type: git
    url: https://github.com/eProsima/Fast-RTPS.git
    version: b48ce9d2fba6fc94e756da01c58b72f2ad848238
  osrf/osrf_testing_tools_cpp:
    type: git
    url: https://github.com/osrf/osrf_testing_tools_cpp.git
    version: 1.1.0
  ros/class_loader:
    type: git
    url: https://github.com/ros/class_loader.git
    version: 1.2.0
  ros/pluginlib:
    type: git
    url: https://github.com/ros/pluginlib.git
    version: 2.2.1
  ros/resource_retriever:
    type: git
    url: https://github.com/ros/resource_retriever.git
    version: 2.1.0
  ros/ros_environment:
    type: git
    url: https://github.com/ros/ros_environment.git
    version: 2.2.0
  ros/urdfdom_headers:
[... 259 lines cut ...]
#### Steps to reproduce issue
I initially posted this on ROS answers: https://answers.ros.org/question/313378/ros2-service-send-request-does-not-send/

But after digging into it some more I have become fairly convinced this is a bug. 

This example uses ros2 services. The server manages a vector of ints. Clients can "register" new values, and the server will append these values to the vector. Clients can also query a position in the vector, and if the position exists, the server will return the value in that position.

The issue I have is that when starting with my launch file (ros2 launch ros_service_mwe mwe_service.launch.py) sometimes the client that is waiting for a value at particular position gets stuck in spin_until_future_complete(). I did some debugging with WireShark and it looks like the service request never makes it to the wire. This is tricky to troubleshoot as it fails ~1 in 10 times on my particular machine, but the failure rate changes depending on the machine.

The easiest way to detect this condition is to look at the terminal output and wait for an instance where the [getNode] does not print: [INFO] [getNode]: Value of Pos: 7

I can solve this issue by adding a timeout / retry strategy to the spin_until_future_complete() call, but I really don't understand the root cause of the issue and am afraid I will mask an issue that will manifest in the future.

I started this minimal example with the add_two_ints example, so the print_usage() function still matches that example. 

Here is a link to the MWE: https://github.com/borgmanJeremy/ros_service_mwe

#### Expected behavior
The expected behavior is that spin_until_future_complete() will return once the server is available to handle the callback.

#### Actual behavior
The actual behavior is spin_until_future_complete() never returns

#### Additional Info:
This might be related to this question? https://github.com/ros2/rclcpp/issues/455

DensoADAS, 2019-02-06:
We encountered the same problem. We'd be glad about a solution.

dschugk, 2019-02-06:
We encountered a similar problem. A solution would be great.

dirk-thomas, 2019-02-06:
The problem comes down to a race condition between (A) when it is detected that the server is running and (B) when the topics between the server and client have actually been matched. In the Crystal release the `client->wait_for_service()` call returns after (A). Hence your first service call races with the event (B). If the call happens before (B) it won't arrive at the server and therefore the getter node won't be able to retrieve the value (ever).

Since the Crystal release [this patch](https://github.com/ros2/rmw_fastrtps/pull/238) has been merged which makes the `wait_for_service` call only return true after (B) has happened. That should avoid the race you are seeing.

Please consider trying to build the latest state from the master branch to check if it resolves your problem.

borgmanJeremy, 2019-02-11:
Thanks - I can confirm that this patch on master fixes my issue.

closed 2019-02-11 by borgmanJeremy