ros2/rclcpp #627: rclcpp/srv/mock.hpp not found - wont build

Jopand, 2019-02-08:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - from source
- Version or commit hash:
  - crystal branch
- DDS implementation:
  - Default - just checked out the code according to the guide and building it
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
"colcon build --packages-select rclcpp"

Unit test is failing:
"test_externally_defined_services.cpp:27:31: fatal error: rclcpp/srv/mock.hpp: No such file or directory
compilation terminated."

The mock.hpp is generated, but it is not being referenced properly, so guess something is missing in the CMakeLists.txt file

#### Expected behavior
Test should pass - project should compile
#### Actual behavior
Project does not compile
#### Additional information
I've been backtracking to see if it's any new changes causing the issue. My guess is that the ament_cmake change from 01/29/2019 is causing the issue (enabling BUILD_TESTING option)
----
## Feature request

#### Feature description

#### Implementation considerations

mjcarroll, 2019-02-12:
I recently ran into this, and it was from a version mismatch. When you say "crystal branch", what versions/actions did you specifically use to check out the source?

mjcarroll, 2019-02-22:
@Jopand I haven't heard back from you.  Were you able to resolve this or are you still having issues?

Jopand, 2019-02-22:
I checked out the latest version tag of all components and the problem is gone (before, I checked out the newest code available in the crystal branch). So it must have been a version mismatch

closed 2019-02-22 by Jopand