ros2/rclcpp #631: Make default intra process manager process dependent instead of context dependent

DensoADAS, 2019-02-14:
Currently the creation of the default intra process manager (as sub context) by "create_default_impl" is done for each context individually. Therefore, the intra process communication within one process, but multiple contexts is not possible (as subscriber/publisher does not find each other).
[rclcpp_patch_static_intra_process_manager.patch.txt](https://github.com/ros2/rclcpp/files/2863423/rclcpp_patch_static_intra_process_manager.patch.txt)

Our proposal (see patch) would be to make the default intra process manager static (-> process dependent) like the default context. So all context share the same manager.

In our case, we create for each composition node an extra context to be able to specify arguments and so on individually. Here we encountered the problem.

wjwwood, 2019-02-14:
Having a separate intra process manager per context is intentional, otherwise it would be impossible to have nodes in the same process which are isolated from one another (especially ones in different domains).

I'm inclined to close this as a "won't fix" unless there's a different rationale of which I'm not aware.

Why can you not share a context between nodes? All the arguments should be able to be targeted to individual nodes without having separate context.

wjwwood, 2019-02-21:
I'm going to close this because I believe the issue has been addressed. Feel free to continue to comment and I can reopen if necessary.

closed 2019-02-21 by wjwwood