ros2/rclcpp #632: Initialize node after construction

lucasw, 2019-02-16:
I'd like to be able to initialize a node after construction in order to set node names and namespaces and anything else after creating an instance with class loader ( https://answers.ros.org/question/315283/set-node-name-namespace-remappings-after-class_loader-createinstance/ ).

I've already created a Node() constructor that doesn't do anything, and an init functions that duplicate the fleshed out constructors  https://github.com/lucasw/rclcpp/tree/node_init .  I imagine other than the one executor add_node throw that checks for a nullptr there is a lot else to add and test to accommodate this change.  Or is there another solution that makes this unnecessary? 

Some related rclcpp development issues I could use help with https://answers.ros.org/question/315794/precedence-of-colcon_ws-packages-over-optroscrystal-versions/ .

lucasw/ros2_cpp_py#5

wjwwood, 2019-02-21:
I don't think creating an instance of `rclcpp::Node` and then setting the node name and namespace is a use case we want to support.

For loading via class loader, I believe the plan is to have a node factory class which has a trivial constructor, and a method that instantiates Node instances (and takes arguments like node name, namespace, and the new `rclcpp::NodeOptions` structure.

See: https://github.com/ros2/launch/issues/160

Sorry, I know you've had some other questions surrounding this and our intentions surrounding it haven't always been well documented.

My inclination is to label this "wont fix" and close it, what do you think @lucasw?

dirk-thomas, 2019-02-21:
While not strictly related to initialization we do want to support changing the node name / namespace later at runtime. So even though the initialization needs to happen at construction time it should be possible to change those in a later init call (at some point in the future when that feature gets implemented).

wjwwood, 2019-02-21:
Yeah, to be clear, I meant we don't want to support Node's which may or may not have their name/namespace set yet (incomplete construction), not that name/namespace should be immutable forever, though they will be immutable for now.

lucasw, 2019-02-21:
I'm fine maintaining my branch and converting when equivalent functionality is available (dashing or later?), ok to close this.

wjwwood, 2019-02-21:
Ok great.

closed 2019-02-21 by wjwwood