ros2/rclcpp #640: std::unique_ptr<> as input parameter instead of std::unique_ptr<>& in publish function

thongpv3, 2019-02-27:
Why don't use `std::unique_ptr<>` as input parameter instead of `std::unique_ptr<>&` in this function:

https://github.com/ros2/rclcpp/blob/8743bcb0a1ef06d6eea5ebbd247e019f9dbb1e10/rclcpp/include/rclcpp/publisher.hpp#L216-L253

I think it is strange because the the parameter is input parameter, not an input and output parameter. So we can use `std::unique_ptr<>` instead. Here is the guideline: https://herbsutter.com/2013/06/05/gotw-91-solution-smart-pointer-parameters/

sloretz, 2019-02-27:
I don't know the reason. Maybe it was added to avoid needing to type `std::move()`?

```c++
#include <memory>

void unique_func_ref(std::unique_ptr<int> & msg) {}
void unique_func_value(std::unique_ptr<int> msg) {}

int main(int argc, char ** argv)
{
   std::unique_ptr<int> foo(new int);
   std::unique_ptr<int> bar(new int);

   unique_func_ref(foo);
   // does not compile
   // unique_func_value(bar);
   unique_func_value(std::move(bar));

   return 0;
}
```

thongpv3, 2019-02-27:
The `std::unique_ptr` intend to be used like that. Anytime users want to pass the ownership, they should explicitly pass std::move, then they would know that this pointer is not valid to use.

thongpv3, 2019-02-27:
And I think this block of code also have a problem:
https://github.com/ros2/rclcpp/blob/8743bcb0a1ef06d6eea5ebbd247e019f9dbb1e10/rclcpp/include/rclcpp/publisher.hpp#L255-L306

We can just provide `virtual void publish(const MessageT & msg)` because we don't managed `std::shared_ptr`'s object lifetime. User can use this like `publish(*shared_ptr_to_object)`. There also increase performance because we don't need to copy a `std::shared_ptr`. And no need to copy paste things :)

sloretz, 2019-03-19:
@thongpv3 would you be willing to make a PR with the `&` removed?

wjwwood, 2019-05-22:
I think this has been resolved, see: https://github.com/ros2/rclcpp/pull/676#issuecomment-489381722

closed 2019-05-22 by wjwwood