ros2/rclcpp #641: Lifecycle nodes don't use simulation time when use_sim_time set

mjeronimo, 2019-02-27:
## Bug report

**Required Info:**

- Operating System: Ubuntu 18.04
- Installation type: source
- Version or commit hash: 10d7b7c72bc243e4e6da0546f9d17e39460df55c
- DDS implementation: Fast-RTPS
- Client library (if applicable): rclcpp

#### Steps to reproduce issue
* Set use_sim_time parameter on a LifecycleNode 
* Later, in a message callback, for example, print out the value of node->now()

#### Expected behavior
The time would be relative to the simulation clock

#### Actual behavior
It is relative to the system clock

mjeronimo, 2019-02-27:
@Karsten1987 No, not related to #642.   Yes, I'll come up with an example for you.

mjeronimo, 2019-03-02:
@Karsten1987 

* Use the lifecycle demo from https://github.com/ros2/demos/tree/master/lifecycle
* Replace lifecycle_talker.cpp with the attached file and build
* Launch gazebo. For example, I used: gazebo --verbose -s libgazebo_ros_init.so ~/src/turtlebot3/src/turtlebot3_simulations/turtlebot3_gazebo/worlds/turtlebot3.world
* Run the lifecycle demo as normal

Once the lifecycle node is activated, the output shows that the lifecycle node's time is the system time, while the normal node's time is relative to /clock. Both nodes have use_sim_time set (see the code). 

[lifecycle_talker.cpp.tar.gz](https://github.com/ros2/rclcpp/files/2921467/lifecycle_talker.cpp.tar.gz)

Karsten1987, 2019-03-26:
with #651 being merged can this issue be closed?

mjeronimo, 2019-03-27:
@Karsten1987 Yes, that one fixes it and this issue can be closed.

closed 2019-03-27 by tfoote