ros2/rclcpp #642: Lifecycle node class is missing parameter-related templates

mjeronimo, 2019-02-27:
## Feature request

#### Feature description

The LifecycleNode class is missing several parameter-related templates that are present in the regular Node class, such as get_parameter_or, get_parameter_or_set, and set_parameter_if_not_set. Adding the missing templates would allow a lifecycle node to be equivalent to normal nodes in its parameter-handling functionality, making it easier to port from a Node to a LifecycleNode.  

#### Implementation considerations

Should be able to copy/paste the templates from the Node header.

Karsten1987, 2019-02-27:
thanks for reporting. The implementation considerations make sense to me given that the lifecycle node has a lot of function forwarding already.

jacobperron, 2019-05-09:
@Karsten1987 I think this has been resolved by #645. Please re-open if I'm mistaken.

commit 81489f77 (2019-03-14): Fixes #642
closed 2019-05-09 by jacobperron