ros2/rclcpp #643: get_node_names should include namespace

wjwwood, 2019-02-28:
`rcl` returns the node names and their namespaces, but in rclcpp you cannot get the node namespaces, see:

https://github.com/ros2/rclcpp/blob/3919ab189795fbf47b5427add06f45d8a43f81ce/rclcpp/src/rclcpp/node_interfaces/node_graph.cpp#L133-L143

There should probably be a new function that returns the nodes and their namespaces separately, and then keep the existing one for convenience but change it to automatically concatenate the name and namespace, because that's likely what you want in most cases.

oswinso, 2019-03-25:
I'd like to work on this. Do you mean like changing `NodeGraph::get_node_names()` to return something like `"/namespace/nodename"`, and then having a separate function that returns a `std::pair` of nodes and their namespaces? But isn't that what `NodeGraph::get_node_names_and_namespaces()` is doing?

mjcarroll, 2019-03-25:
For consistency with: https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/node_interfaces/node_base.hpp#L49

You could probably do:

```
std::vector<std::pair<std::string, std::string>> get_node_names_and_namespaces();
std::vector<std::string> get_node_names();
std::vector<std::string> get_node_namespaces();
std::vector<std::string> get_node_fully_qualified_names()
```

Is that what you were thinking @wjwwood ?

wjwwood, 2019-03-25:
Yes, except I don't think `get_node_namespaces()` is needed.

wjwwood, 2019-03-25:
Well, and I was advocating for just changing `get_node_names()` to return the combined namespace/name, so then `get_node_fully_qualified_names()` is redundant. But that would break "ABI", in the behavior sense, so it would have to go into Dashing only.

jhdcs, 2019-03-26:
I've also got a possible solution, assuming I've got my environment set up correctly. Would you like me to submit a pull request as well? Also, should we write tests for this functionality?

wjwwood, 2019-03-26:
> Would you like me to submit a pull request as well?

Ideally you guys could collaborate and avoid duplicated effort. I think the best way to do that is to open a pull request as early as possible, so yes I think that's a good idea.

> Also, should we write tests for this functionality?

Always :)

jhdcs, 2019-03-26:
Having a few issues creating a pull request. Should I clone the repository to my GitHub account in addition to my local computer or something?

wjwwood, 2019-03-26:
Yes, GitHub has pretty good documentation on contributing via a pull request:

https://help.github.com/en/articles/creating-a-pull-request

oswinso, 2019-03-26:
I'm presuming that @jhdcs has already done the work, and I shouldn't make a PR request?

wjwwood, 2019-03-26:
I think there's work to be done yet, maybe you guys could collaborate? I already made some comments on his pr, and it still needs API documentation, tests, and perhaps external documentation (like in the migration notes for dashing: https://index.ros.org/doc/ros2/Releases/Release-Dashing-Diademata/#changes-since-the-crystal-release).

wjwwood, 2019-03-26:
@jhdcs's pr is here btw: https://github.com/ros2/rclcpp/pull/669

jhdcs, 2019-03-27:
Agreed. This is just a work in progress. And an extra pair of eyes is always helpful.

Is there somewhere you'd like to work on?

oswinso, 2019-03-27:
I'm free to work on anything, but I'm not sure which parts you are available to work on, as well as how I could help considering its on your fork.

jhdcs, 2019-03-27:
Sorry for the late reply, re-working on getting CLion up and running. Still, you should be a collaborator on the fork now.

You're welcome to work on anything. I was planning on refactoring the functions to align with @wjwwood's requests once I get CLion working again, but if you want to do that, I don't mind. We also need to make tests. And once I figure out the proper area for API documentation, I'll start on that.

oswinso, 2019-03-27:
Ok cool. If you don't mind, I just pushed some changes that aligned with what @wjwwood was thinking this should look like.

jhdcs, 2019-04-01:
I think I've _mostly_ gotten my environment to work again. Change looked pretty good. I only removed an unnecessary capture-by-reference in the lambda.
@wjwwood, would you like us to modify test_node.cpp to implement our tests? Or create a new file?

wjwwood, 2019-04-01:
> @wjwwood, would you like us to modify test_node.cpp to implement our tests? Or create a new file?

Up to you, I prefer keeping them in the same file, or failing that, making a folder `test_node/<each test>.cpp` so they are organized together.

jhdcs, 2019-04-05:
@wjwwood What's left on this? Just the API documentation? That's on the wiki, right?

wjwwood, 2019-04-05:
The pr needs passing CI, and API documentation should be done in the `node.hpp` header directly.

wjwwood, 2020-03-23:
Closing with https://github.com/ros2/rclcpp/pull/698

closed 2020-03-23 by wjwwood