ros2/rclcpp #647: change throw_from_rcl_error to exception_from_rcl_error

sloretz, 2019-03-04:
## Feature request

Currently rclcpp has a function to throw an exception based on an rcl error. 

https://github.com/ros2/rclcpp/blob/3919ab189795fbf47b5427add06f45d8a43f81ce/rclcpp/src/rclcpp/clock.cpp#L134

This is a request to change that function to return an exception without throwing.

```c++
throw rclcpp::exceptions::from_rcl_error(ret, "Failed to add time jump callback");
```
I think this could help static analysis tools like `cppcheck` recognize when code won't be executed due to an exception. This would remove the need for ros2/rclcpp#646

#### Implementation considerations
`throw_from_rcl_error` could be refactored to use `from_rcl_error` rather than changing all the code at once.

wjwwood, 2019-03-05:
Seems reasonable, at the very least to factor the creation of the exception and the throw into separate functions.

jhdcs, 2019-03-26:
Mind if I give this a go? I was thinking of renaming "throw_from_rcl_error" to either "from_rcl_error" or "gen_rcl_exception", then refactoring everything that called the original function to "throw <from_rcl_error>". Unless, of course, you want to do something besides throwing at any of the locations.

Also, out of curiosity, is there a reason we aren't using enums for return codes? And should I put that in a separate question?

wjwwood, 2019-03-26:
> Mind if I give this a go?

Sure, please have a go at it.

> I was thinking of renaming "throw_from_rcl_error" to either "from_rcl_error" or "gen_rcl_exception", then refactoring everything that called the original function to "throw <from_rcl_error>". Unless, of course, you want to do something besides throwing at any of the locations.

I think `from_rcl_error` is better than `gen_rcl_exception` myself, but `exception_from_rcl_error` would be another alternative that I think. Though in context, `throw rclcpp::exceptions::from_rcl_error` is pretty clear already. On the other hand, if the user does `using rclcpp::exceptions::from_rcl_error` and later does `throw from_rcl_error(...)` it's a bit less clear and `throw exception_from_rcl_error(...)` would be clearer in that case.

I don't mind which you do, but the already proposed ones seem like the best name I've heard so far.

Also, consider the line from the description:

> `throw_from_rcl_error` could be refactored to use `from_rcl_error` rather than changing all the code at once.

Avoiding touching all the existing code would be a nice thing to have, while still enabling users to generate the exception without throwing it in cases where that's needed.

----

> Also, out of curiosity, is there a reason we aren't using enums for return codes? And should I put that in a separate question?

I think it's mostly a historical reason, but avoiding enums in C (and using `#define` instead) forces us to pick a value for each constant, whereas enums would let us list them and automatically get values from their order, but then later we'd be encouraged to reorder them or inject new ones between existing ones, causing the values to change. By using the defines we have to be explicit about the values, and plan for future ones, reducing the likelihood that we need to change the value of an existing return code.

sloretz, 2019-04-02:
I think this ticket is invalid as written. I assumed `from_rcl_error` could return a base exception type, but I didn't realize that only triggers the copy constructor of the base type. The derived exception type would be lost and `catch (DerivedType &)` would never trigger.

The original issue is having to work around cppcheck's false positive. I think a better solution would would be for `cppcheck` to understand [`[[noreturn]]`](https://en.cppreference.com/w/cpp/language/attributes/noreturn) and then to mark `throw_from_rcl_error` with that.

Another option would be to add a cppcheck library config file for rclcpp that has `<noreturn>` set for `throw_from_rcl_error`, but that requires changes to `ament_cppcheck` and `ament_cmake_cppcheck`. They would have to take config files, and then downstream libraries would need to be able to get cppcheck config files from upstream ones. That seems like more effort than occasional PRs to work around false positives.

sloretz, 2019-04-16:
Closing now that #678 is merged

closed 2019-04-16 by sloretz