ros2/rclcpp #668: Subscription count and intra-process subscription counts are not updated in sync

cvasfi, 2019-03-26:
## Bug report

Intra-process subscription count of Publisher is not updated in sync with the subscription count when intra-process communication is used.
 When the `get_subscription_count()` method returns the updated count,  `get_intra_process_subscription_count()` may not return the updated count yet. This behaviour causes the unit tests in [test_publisher_subscription_count_api](https://github.com/ros2/rclcpp/blob/master/rclcpp/test/test_publisher_subscription_count_api.cpp) to fail [occasionally](https://ci.ros2.org/view/nightly/job/nightly_linux_repeated/1323/testReport/(root)/projectroot/test_publisher_subscription_count_api/).

**Required Info:**

- Operating System:
  - Ubuntu 16.04 
- Installation type:
  - Source 
- Version or commit hash:
  - Crystal 
- DDS implementation:
  - Fast-RTPS 
- Client library (if applicable):
  - rclcpp 

#### Steps to reproduce issue
Repeat the unit test [test_publisher_subscription_count_api](https://github.com/ros2/rclcpp/blob/master/rclcpp/test/test_publisher_subscription_count_api.cpp) until failure. Though this may be hard to reproduce locally, here's an instance of such a failure:  https://ci.ros2.org/view/nightly/job/nightly_linux_repeated/1323/testReport/(root)/projectroot/test_publisher_subscription_count_api/

Above test logs shows that the equality conditions on `get_subscription_count()`  are satisfied yet the conditions on  `get_intra_process_subscription_count()` are not met.

#### Expected behavior
`get_subscription_count()` and `get_intra_process_subscription_count()` methods should be in sync when intra-process communication is used.

#### Actual behavior
The methods return different results.

#### Additional information

wjwwood, 2019-03-26:
> `get_subscription_count()` and `get_intra_process_subscription_count()` methods should be in sync when intra-process communication is used.

That's not possible, they are two separate subscriptions (two separate DataWriters under the hood if using DDS).

As currently stated, the expected behavior you put forward is not possible and would be a won't fix.

There might other ways to address the symptom ("This behaviour causes the unit tests in test_publisher_subscription_count_api to fail occasionally."), but making them update in sync is, in my opinion, not the correct way to fix the problem, nor is it possible to implement right now.

ivanpauno, 2019-03-27:
I agree with @wjwwood, that syncing them is not a good idea.

But it shouldn't fail with get_intra_process_subscription_count, as this is not being asked to the underlying DDS "datawriter". It's obtained from the size of an internal map in the intraprocess manager, which is updated when creating the publisher and when deleting it.
https://github.com/ros2/rclcpp/blob/b352d450319e84dd28aed78b800ab40938cec5c2/rclcpp/include/rclcpp/intra_process_manager_impl.hpp#L254-L272

I will check later if I can repro this, and see if I can understand why sometimes the count is not being updated.

wjwwood, 2019-03-27:
@ivanpauno I assumed we were asking the underlying datawriter on the intra process topic. Based on what you said, my theory no longer makes sense to me...

I would appreciate it if you could look into @ivanpauno (considering you also wrote the test, you might have unique insight).

ivanpauno, 2019-03-28:
> @ivanpauno I assumed we were asking the underlying datawriter on the intra process topic.

We aren't doing that, because it would return a wrong count in the case of two process with publishers and subscriber in the same topic using intraprocess comm.

> I would appreciate it if you could look into @ivanpauno (considering you also wrote the test, you might have unique insight).

I've already reproduced the issue. I'm trying to figure out what the problem is.

PR #671 (2019-04-04): Replace const char * with a std::array<char, TOPIC_NAME_LENGTH> as the key of IPM IDTopicMap
Closes #668

## Changes

Replaces the `const char *` key in IDTopicMap with a std::array<char, TOPIC_NAME_LENGTH> key. A copy of the string is forced.

## Problem description

After some debugging, I realized what this random error was.
Sometimes, the topic name  wasn't found in the subscription_ids_by_topic_ map here:
https://github.com/ros2/rclcpp/blob/cb20529e5e218cde0106c59b12067a9ca9f43022/rclcpp/include/rclcpp/intra_process_manager_impl.hpp#L266-L270
And a 0 subscription count was returned.

The  [... cut]
closed 2019-04-04 by ivanpauno