ros2/rclcpp #679: logging macros will not take a const Logger

wjwwood, 2019-04-02:
Right now if you try to pass a `const rclcpp::Logger` to a logging macro you'll get an error like this (taken from https://github.com/ros-planning/moveit2/issues/21#issuecomment-478259602):

```
/Users/victor/ros2_ws/install/rclcpp/include/rclcpp/logging.hpp:411:3: note: expanded from macro 'RCLCPP_ERROR'
  static_assert( \
  ^
/Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/robot_state/src/conversions.cpp:100:9: error: static_assert failed due to requirement '::std::is_same<typename std::remove_reference<decltype(LOGGER)>::type, typename ::rclcpp::Logger>::value' "First argument to logging macros must be an rclcpp::Logger"
        RCLCPP_ERROR(LOGGER, "Caught %s", ex.what());
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

We seem to remove the reference but not the const qualifier when comparing with `std::is_same`.

Karsten1987, 2019-04-25:
Can this be closed now that https://github.com/ros2/rclcpp/pull/680 is merged?

wjwwood, 2019-04-25:
Yup.

Karsten1987, 2019-08-08:
I believe this is still not correctly solved:

```
bool log_function(const rclcpp::Logger & logger)
{
  RCLCPP_INFO(logger, "successful log");
  return true;
}

TEST_F(TestLoggingMacros, test_log_from_node) {
  auto logger = rclcpp::get_logger("test_logging_logger");
  bool success = log_function(logger);
  EXPECT_TRUE(success);
}
```

yields to

```
/Users/karsten/workspace/osrf/ros2_full/src/ros2/rclcpp/rclcpp/test/test_logging.cpp:168:3: error: static_assert failed due to requirement '::std::is_same<typename std::remove_reference<typename std::remove_cv<decltype(logger)>::type>::type, typename ::rclcpp::Logger>::value' "First argument to logging macros must be an rclcpp::Logger"
  RCLCPP_INFO(logger, "successful log");
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/karsten/workspace/osrf/ros2_full/build/rclcpp/include/rclcpp/logging.hpp:214:5: note: expanded from macro 'RCLCPP_INFO'
    static_assert( \
    ^
1 error generated.
```

emersonknapp, 2019-08-23:
@Karsten1987 was this issue resolved by https://github.com/ros2/rclcpp/pull/820 ?

closed 2019-08-23 by Karsten1987