ros2/rclcpp #687: Avoid comparing strings in IntraProcessManagerImpl

ivanpauno, 2019-04-08:
## Feature request

#### Feature description
I've taken the idea from [this comment](https://github.com/ros2/rclcpp/pull/671#discussion_r271858293). Basically, we are now comparing strings when storing a message in the Intra Process Manager (IPM), which is not good at all from a performance point of view.

https://github.com/ros2/rclcpp/blob/af9ae4a61c24ba207f3db6a32c41e0e52cb29cb9/rclcpp/include/rclcpp/intra_process_manager_impl.hpp#L180-L181

https://github.com/ros2/rclcpp/blob/af9ae4a61c24ba207f3db6a32c41e0e52cb29cb9/rclcpp/include/rclcpp/intra_process_manager_impl.hpp#L281-L301

https://github.com/ros2/rclcpp/blob/af9ae4a61c24ba207f3db6a32c41e0e52cb29cb9/rclcpp/include/rclcpp/intra_process_manager_impl.hpp#L314-L322

#### Implementation considerations
It could be easily solved in two ways:
- [1] As has been already proposed by @hidmic, use an unordered map. It will be needed to add a custom hash function, as cpp standard library doesn't provide a "free" hash function for strings. It only provides a hash function for std::string, and we are using `std::array<char, N> (check why [here](https://github.com/ros2/rclcpp/pull/671)).
- [2] We could already have a subscription map ordered by publisher id before store_intra_process_message is called (which was suggested by @wjwwood). This is what I have in mind:
  - All the existing maps will be kept.
  - Add another map called `publisher_ids_by_topic_`. When a publisher is added to the ipm, is organized in this map.
  - Add another map called `subscription_ids_by_publisher_`. When a subscription is added, the `publishers_id_by_topic` map is queried, and with that info the subscription id is added to all the corresponding sets in the `subscription_ids_by_publisher` map. It will be also stored in `subscription_ids_by_topic_` map. When a publisher is added, the `subscription_ids_by_publisher_` map has to be filled with the existing data in `subscription_ids_by_topic_`.

Option [2] is kind of redundant, but that's not a big deal. Maybe, it can be reorganized in a better way.
@wjwwood @hidmic Do you have any opinions in which path to take?

wjwwood, 2019-04-24:
Option 2 sounds best (clearly I'm biased, lol), and it is redundant, but it's one of those classic memory for computation trade-offs. I personally think the additional memory used will be acceptable.

wjwwood, 2020-03-23:
@ivanpauno does this still apply since we refactored intra-process in Eloquent?

ivanpauno, 2020-03-23:
> @ivanpauno does this still apply since we refactored intra-process in Eloquent?

No, it doesn't. Closing!

closed 2020-03-23 by ivanpauno