ros2/rclcpp #706: Make add_to_wait_set() throw

mm318, 2019-04-26:
## Feature request

#### Feature description
Currently, some implementations of `Waitable::add_to_wait_set()` do not throw exceptions and only return a `false` on failure instead. It was suggested that they should, and also since their counterparts in the `rclpy` do throw exceptions.

#### Implementation considerations
Some functions found in this repo that needs this enhancement include
```
ServerBase::add_to_wait_set()
ClientBase::add_to_wait_set()
```
May also consider converting the use of `RCUTILS_LOG_ERROR_NAMED()` in `AllocatorMemoryStrategy::add_handles_to_wait_set()` to throws as well.

mm318, 2019-04-26:
The original discussion started here: https://github.com/ros2/rclcpp/pull/695#discussion_r278328389

CursedRock17, 2023-06-12:
As of currently, it seems that
>
```
ServerBase::add_to_wait_set()
ClientBase::add_to_wait_set()
```
have both been correctly implemented.

I know it has been 4 years since these changes were proposed, so the [original file](https://github.com/aws-ros-dev/rclcpp/blob/763963a2a19561a8282a10523102738f8dbfd166/rclcpp/src/rclcpp/qos_event.cpp) in the aws PR has changed to the [current version](https://github.com/aws-ros-dev/rclcpp/blob/763963a2a19561a8282a10523102738f8dbfd166/rclcpp/src/rclcpp/qos_event.cpp) which removes the `return false` after throwing the exception.

Then, the two functions this issue talks about on both [client](https://github.com/ros2/rclcpp/blob/b3518d12ca2a123174384a740a6430f3c1660448/rclcpp_action/src/client.cpp#L261) and the [server](https://github.com/ros2/rclcpp/blob/b3518d12ca2a123174384a740a6430f3c1660448/rclcpp_action/src/server.cpp#L174), have also both been updated to throw an exception, but since their return type was `void`, they don't return true afterwards. Unless files which currently don't worry about throwing at all like [SubscriptionIntraProcessBase](https://github.com/ros2/rclcpp/blob/b3518d12ca2a123174384a740a6430f3c1660448/rclcpp/src/rclcpp/subscription_intra_process_base.cpp#L21) or [StaticExecutorEntitiesCollector](https://github.com/ros2/rclcpp/blob/b3518d12ca2a123174384a740a6430f3c1660448/rclcpp/src/rclcpp/executors/static_executor_entities_collector.cpp#L271) are supposed to be changed around. Should issue be sealed similar to the original?

clalancette, 2024-01-19:
Given that the original issue ticketed here is fixed, I'm going to close this out.

closed 2024-01-19 by clalancette