ros2/rclcpp #717: Some times, callback are not called during a call to Executor::spin_util_future_complete

adrienbarral, 2019-05-08:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Binaries
- Version or commit hash:
  - rclcpp version ros-crystal-rclcpp : 0.6.4-0bionic.20190406.051903
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
You can try this unit test file :

```
#include <chrono>
#include <gtest/gtest.h>
#include <rclcpp/executors.hpp>
#include <rclcpp/node.hpp>
#include <std_msgs/msg/float32.hpp>

using namespace std::literals;
using namespace rclcpp;
using namespace std_msgs::msg;
using namespace rclcpp::executor;

bool wait_for_subscriber(std::shared_ptr<rclcpp::Node> node,
                         const std::string& topic_name, std::chrono::milliseconds timeout,
                         std::chrono::microseconds sleep_period)
{
    using std::chrono::duration_cast;
    using std::chrono::microseconds;
    using std::chrono::steady_clock;
    auto start = steady_clock::now();
    microseconds time_slept(0);
    auto predicate = [&node, &topic_name]() -> bool {
        // the subscriber is available if the count is gt 0
        return node->count_subscribers(topic_name) > 0;
    };
    while(!predicate() && time_slept < duration_cast<microseconds>(timeout))
    {
        rclcpp::Event::SharedPtr graph_event = node->get_graph_event();
        node->wait_for_graph_change(graph_event, sleep_period);
        time_slept =
            duration_cast<std::chrono::microseconds>(steady_clock::now() - start);
    }
    int64_t time_slept_count =
        std::chrono::duration_cast<std::chrono::nanoseconds>(time_slept).count();
    RCLCPP_INFO(node->get_logger(),
                "Waited %d nanoseconds for the subscriber to connect to topic '%s'. Sub "
                "count : %d\n",
                time_slept_count, topic_name.c_str(),
                node->count_subscribers(topic_name));

    return !predicate();
}

class Filter : public Node
{
public:
    Filter() : Node("filter", "", true)
    {
        filteredDataPub = create_publisher<Float32>("filtered");
        dataIn = create_subscription<Float32>("dataIn", [this](Float32::SharedPtr in) {
            RCLCPP_INFO(get_logger(), "Data In callback called");
            auto res = std::make_shared<Float32>();
            res->data = in->data * 2;
            filteredDataPub->publish(res);
        });
    }

protected:
    Publisher<Float32>::SharedPtr filteredDataPub;
    Subscription<Float32>::SharedPtr dataIn;
};

class TesterNode : public Node
{
public:
    TesterNode() : Node("tester", "", true)
    {
        filteredSubscriber =
            create_subscription<Float32>("filtered", [this](Float32::SharedPtr in) {
                RCLCPP_INFO(get_logger(), "Filtered callback called");
                filteredPromise.set_value(in->data);
            });
        dataInPublisher = create_publisher<Float32>("dataIn");
    }
    void publishData(float data)
    {
        auto res = std::make_shared<Float32>();
        res->data = data;
        dataInPublisher->publish(res);
    }
    std::promise<float> filteredPromise;

protected:
    rclcpp::Publisher<Float32>::SharedPtr dataInPublisher;
    rclcpp::Subscription<Float32>::SharedPtr filteredSubscriber;
};

class TestFixture : public ::testing::Test
{
protected:
    void SetUp()
    {

        testerNode = std::make_shared<TesterNode>();
        ASSERT_EQ(testerNode->count_subscribers("dataIn"), static_cast<size_t>(0));
        filterNode = std::make_shared<Filter>();

        executor.add_node(filterNode);
        executor.add_node(testerNode);

        ASSERT_FALSE(wait_for_subscriber(testerNode, "dataIn", 1000ms, 5ms));
    }

    void TearDown()
    {
        executor.cancel();
        executor.remove_node(filterNode);
        executor.remove_node(testerNode);
        testerNode.reset();
        filterNode.reset();
    }

    std::shared_ptr<TesterNode> testerNode;
    rclcpp::executors::SingleThreadedExecutor executor;

private:
    std::shared_ptr<Filter> filterNode;
};

TEST_F(TestFixture, weCanFilterOurDatas)
{
    auto future = std::shared_future<float>(testerNode->filteredPromise.get_future());
    testerNode->publishData(5.0);
    ASSERT_EQ(this->executor.spin_until_future_complete(future, std::chrono::seconds(1)),
              FutureReturnCode::SUCCESS);
    ASSERT_EQ(future.get(), 10.);
}

int main(int argc, char** argv)
{
    rclcpp::init(argc, argv);
    ::testing::InitGoogleTest(&argc, argv);
    auto res = RUN_ALL_TESTS();
    rclcpp::shutdown();
    return res;
}
```

#### Expected behavior
I excepect that this test pass every time when I launch it.

#### Actual behavior
Some times, this test doesn't pass. 
When this test doesn't pass it is always because the spin_util_future_complete return a timeout.

`ASSERT_EQ(this->executor.spin_until_future_complete(future, std::chrono::seconds(1)),
              FutureReturnCode::SUCCESS);`

When this occurs, I can't read neither the dataIn callback trace, nor the filtered trace. So that looks like the callback of "dataIn" subscription is never called.

#### Additional information
If I try to rise the timeout in spin_until_future_complete (to 10 seconds), test will also fail.

I also post this "bug" as question on [Answers.ros.org](https://answers.ros.org/question/322599/how-to-write-a-deterministic-unit-test-with-rclcpp/) because, I am not really sure that is a bug. Maybe I didn't well understood how executors work. But nobody answers there.

borgmanJeremy, 2019-05-10:
I looks like you only publish the message once? If so the message might be broadcast before the publisher has been been matched with a subscriber. You can use a service rather than pub/sub if you want to make sure an event driven message is delivered. 

This thread may be relevant: https://github.com/ros2/ros2/issues/344

adrienbarral, 2019-05-10:
@borgmanJeremy , I publish the message once, but before publishing it I wait for subscriber to be connected to publisher (thanks to the call to method `wait_for_subscriber(testerNode, "dataIn", 1000ms, 5ms)` ).

I think there is a big difference between my ticket and the one you linked. In my issue I am blocking execution of the thread until callback is called thanks to the `spin_util_future_complete`. In the issue you cite, there is a second thread in which `spin` is running.

wjwwood, 2019-05-10:
> I also post this "bug" as question on Answers.ros.org because, I am not really sure that is a bug. Maybe I didn't well understood how executors work. But nobody answers there.

Sorry, I've seen some of the questions about how the executor works and I've been wanting to relpy, but we've been super busy for the last few months. I had decided I would reply to those questions with links to updated documentation after doing some changes that were planned to the executor's architecture, but we ended up pushing that work off until later, so that also contributed to the delay.

----

I'll attempt to look into this issue in a few weeks, if no one else gets to it first.

1r0b1n0, 2019-05-20:
Hi,
When running this code with `__log_level:=debug` I found some some interesting information :
It seems like sometimes subscriberNode->count_publishers(topic) returns "1" while the subscription isn't really finished yet.
Moreover there was a small error in the line `looking for subscriber topic` that has been fixed in ros2/rmw_fastrtps#279

Sometimes I have this output (fail) :
```
1558371495.682921408: [rmw_fastrtps_shared_cpp] [DEBUG] - looking for subscriber topic: /dataIn, number of matches: 1
1558371495.682989051: [rmw_fastrtps_shared_cpp] [DEBUG] - looking for subscriber topic: /dataIn, number of matches: 0
1558371495.688210422: [rmw_fastrtps_shared_cpp] [DEBUG] - looking for subscriber topic: /dataIn, number of matches: 1
1558371495.688284384: [rmw_fastrtps_shared_cpp] [DEBUG] - looking for subscriber topic: /dataIn, number of matches: 0
1558371495.693456493: [rmw_fastrtps_shared_cpp] [DEBUG] - looking for subscriber topic: /dataIn, number of matches: 1
1558371495.693514447: [rmw_fastrtps_shared_cpp] [DEBUG] - looking for subscriber topic: /dataIn, number of matches: 0
1558371495.693849354: [rmw_fastrtps_shared_cpp] [DEBUG] - Adding topic 'rt/rosout' with type 'rcl_interfaces::msg::dds_::Log_' for node '1.f.74.1.17.5a.0.0.1.0.0.0|0.0.1.c1'
1558371495.694061311: [rmw_fastrtps_shared_cpp] [DEBUG] - Adding topic 'rt/rosout' with type 'rcl_interfaces::msg::dds_::Log_' for node '1.f.74.1.17.5a.0.0.2.0.0.0|0.0.1.c1'
1558371495.694168301: [rcl] [DEBUG] - Guard condition in wait set is ready
1558371495.694232581: [rcl] [DEBUG] - Waiting without timeout
1558371495.694236791: [rmw_fastrtps_shared_cpp] [DEBUG] - Adding topic 'rr/tester/get_parametersReply' with type 'rcl_interfaces::srv::dds_::GetParameters_Response_' for node '1.f.74.1.17.5a.0.0.1.0.0.0|0.0.1.c1'
1558371495.694264232: [rcl] [DEBUG] - Timeout calculated based on next scheduled timer: false
1558371495.694334731: [rmw_fastrtps_shared_cpp] [DEBUG] - looking for subscriber topic: /dataIn, number of matches: 1
1558371495.694390866: [rmw_fastrtps_shared_cpp] [DEBUG] - looking for subscriber topic: /dataIn, number of matches: 0
1558371495.694553416: [rmw_fastrtps_shared_cpp] [DEBUG] - Adding topic 'rr/filter/get_parametersReply' with type 'rcl_interfaces::srv::dds_::GetParameters_Response_' for node '1.f.74.1.17.5a.0.0.2.0.0.0|0.0.1.c1'
1558371495.694586377: [rmw_fastrtps_shared_cpp] [DEBUG] - Adding topic 'rr/tester/get_parameter_typesReply' with type 'rcl_interfaces::srv::dds_::GetParameterTypes_Response_' for node '1.f.74.1.17.5a.0.0.1.0.0.0|0.0.1.c1'
1558371495.694629206: [rcl] [DEBUG] - Guard condition in wait set is ready
1558371495.694675479: [rcl] [DEBUG] - Waiting without timeout
1558371495.694697546: [rcl] [DEBUG] - Timeout calculated based on next scheduled timer: false
1558371495.694766744: [rmw_fastrtps_shared_cpp] [DEBUG] - looking for subscriber topic: /dataIn, number of matches: 1
1558371495.694821207: [rmw_fastrtps_shared_cpp] [DEBUG] - looking for subscriber topic: /dataIn, number of matches: 0
1558371495.694919478: [rmw_fastrtps_shared_cpp] [DEBUG] - Adding topic 'rr/tester/set_parametersReply' with type 'rcl_interfaces::srv::dds_::SetParameters_Response_' for node '1.f.74.1.17.5a.0.0.1.0.0.0|0.0.1.c1'
1558371495.694981986: [rmw_fastrtps_shared_cpp] [DEBUG] - Adding topic 'rr/filter/get_parameter_typesReply' with type 'rcl_interfaces::srv::dds_::GetParameterTypes_Response_' for node '1.f.74.1.17.5a.0.0.2.0.0.0|0.0.1.c1'
1558371495.695050443: [rcl] [DEBUG] - Guard condition in wait set is ready
1558371495.695095095: [rcl] [DEBUG] - Waiting without timeout
1558371495.695118775: [rcl] [DEBUG] - Timeout calculated based on next scheduled timer: false
1558371495.695188656: [rmw_fastrtps_shared_cpp] [DEBUG] - looking for subscriber topic: /dataIn, number of matches: 1
1558371495.695252642: [rmw_fastrtps_shared_cpp] [DEBUG] - Adding topic 'rr/tester/set_parameters_atomicallyReply' with type 'rcl_interfaces::srv::dds_::SetParametersAtomically_Response_' for node '1.f.74.1.17.5a.0.0.1.0.0.0|0.0.1.c1'
1558371495.695356968: [rmw_fastrtps_shared_cpp] [DEBUG] - looking for subscriber topic: /dataIn, number of matches: 0
1558371495.695394768: [rmw_fastrtps_shared_cpp] [DEBUG] - Adding topic 'rr/filter/set_parametersReply' with type 'rcl_interfaces::srv::dds_::SetParameters_Response_' for node '1.f.74.1.17.5a.0.0.2.0.0.0|0.0.1.c1'
1558371495.695461812: [rcl] [DEBUG] - Guard condition in wait set is ready
1558371495.695506004: [rcl] [DEBUG] - Waiting without timeout
1558371495.695528658: [rcl] [DEBUG] - Timeout calculated based on next scheduled timer: false
1558371495.695564931: [rmw_fastrtps_shared_cpp] [DEBUG] - looking for subscriber topic: /dataIn, number of matches: 1
1558371495.695583987: [rmw_fastrtps_shared_cpp] [DEBUG] - Adding topic 'rr/tester/describe_parametersReply' with type 'rcl_interfaces::srv::dds_::DescribeParameters_Response_' for node '1.f.74.1.17.5a.0.0.1.0.0.0|0.0.1.c1'
1558371495.695671770: [rmw_fastrtps_shared_cpp] [DEBUG] - looking for subscriber topic: /dataIn, number of matches: 0
1558371495.695814813: [rmw_fastrtps_shared_cpp] [DEBUG] - Adding topic 'rr/filter/set_parameters_atomicallyReply' with type 'rcl_interfaces::srv::dds_::SetParametersAtomically_Response_' for node '1.f.74.1.17.5a.0.0.2.0.0.0|0.0.1.c1'
1558371495.695882448: [rcl] [DEBUG] - Guard condition in wait set is ready
1558371495.695927673: [rcl] [DEBUG] - Waiting without timeout
[... cut ...]
1558371875.407956914: [rcl] [DEBUG] - Shutting down ROS client library, for context at address: 0x5624767af070
1558371875.408034781: [rcl] [DEBUG] - Guard condition in wait set is ready
1558371875.408115849: [rclcpp] [DEBUG] - SignalHandler::uninstall(): notifying deferred signal handler
1558371875.408147437: [rclcpp] [DEBUG] - deferred_signal_handler(): woken up due to SIGINT or uninstall
1558371875.408177989: [rclcpp] [DEBUG] - deferred_signal_handler(): signal handling uninstalled
1558371875.408236462: [rclcpp] [DEBUG] - signal handler uninstalled
```

adrienbarral, 2019-05-28:
Ok, I think I found what is the problem, and of course, the problem is not in rclcpp, but in my test...

When reading carefully logs in the @1r0b1n0 message, I can see that when test fail, this is because topic **filtered** is  added after we finished waiting for topic **dataIn**.
And when the test pass, **filtered** is added before **dataIn**.

So the problem is just that we must wait for both connections to be established before continuing !!!

Adding the following line in my SetUp function make the test pass every times :

```C++
 ASSERT_FALSE(wait_for_subscriber(testerNode, "dataIn", 1000ms, 5ms));
 ASSERT_FALSE(wait_for_subscriber(filterNode, "filtered", 1000ms, 5ms));
```

closed 2019-05-28 by adrienbarral