ros2/rclcpp #721: document and test rclcpp::Node::list_parameters

wjwwood, 2019-05-15:
As a follow up of https://github.com/ros2/rclcpp/pull/495 (the method has existed for a long time, but I just didn't have time to fix it up in that pr), we need to still document and test the `rclcpp::Node::list_parameters` method.

CursedRock17, 2023-09-25:
It seems that pull request #1342 solved the tests for this function, is there any need for the documentation to be updated?

fujitatomoya, 2023-09-25:
@CursedRock17 i think the rest of the work here is

https://github.com/ros2/rclcpp/blob/9284d7cefa9e37ce3b6926a44e6177d6e4a4b62a/rclcpp/include/rclcpp/node.hpp#L975

CursedRock17, 2023-09-26:
@fujitatomoya I did see that, but it was added in [this commit](https://github.com/ros2/rclcpp/commit/6b108414779af49e22ffa13d1b9cb1fa2763d923#diff-89c3c3860cf77d4493d27c596422e015691a1d314d5a38deca6d6bb5a341106c) in April 2019, whereas the pull request to "fix this" was added in September 2020. That being said, if there's still not enough documentation/testing around list_parameters I can still work on it. Although, there's a whole test [here](https://github.com/ros2/rclcpp/blob/rolling/rclcpp/test/rclcpp/test_node.cpp#L1043-L1054) for `list_parameter` which should be enough.

fujitatomoya, 2023-09-26:
@CursedRock17 i think tests have been added with `list_parameters` but still doc section is not well described? that is what i mean here.

https://github.com/ros2/rclcpp/blob/9284d7cefa9e37ce3b6926a44e6177d6e4a4b62a/rclcpp/include/rclcpp/node.hpp#L975

so adding documentation is the rest of the work here? what do you think?

CursedRock17, 2023-09-26:
That's fair there's not much documentation there, do you mind if I make a pull request for this?

fujitatomoya, 2023-09-26:
Please go for it, happy to do review 👍

closed 2023-09-28 by fujitatomoya